code-explorer1234/Yi-RTS3903N-RTSPServer #2: 7.1.00.11A_201905311440 firmware offline

ruinidavide, 2021-08-08:
Hi thank you for the job.

I've an yi iot camera (mibao camera) with 7.1.00.11A_201905311440 firmware.

I unzip the content of Yi-RTS3903N-RTSPServerV01.tar.gz
on an 8gb sd card formatted in fat32, at the root.

With the camera off, I insert the SD card and then when I turn it on, the rtsp service does not work and the camera appear all time offline.

I've modified wpa_supplicant.conf in /sdcard/Factory/ with your WiFi credentials :

network={
ssid="my ssid "
psk="very secret passphrase"
}

cjj25, 2021-08-08:
My pleasure! Let's see if we can get this working.

Can you match the original layout of the file and replace the exact areas that are SSID_NAME_OF_WIFI and WIFI_SECRET_KEY

Original:
```
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="SSID_NAME_OF_WIFI"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="WIFI_SECRET_KEY"
}
network={
ssid="SSID_NAME_OF_WIFI"
scan_ssid=1
key_mgmt=NONE
wep_key0=WIFI_SECRET_KEY
wep_tx_keyidx=0
}

```

Let me know if you get it connected then? I'm going to be pushing a new update in 5 minutes.

cjj25, 2021-08-08:
New version uploaded - please try using this V02 and report back

cjj25, 2021-08-13:
Any update on this?

fmac83, 2021-08-15:
I have a camera Floureon with the same firmware and rtsp is working fine, the only problem is the pink image, changing invert_adc does nothing, it turns to gray. I cannot find stream.c. 
By the way, I have a question, after starting rtsp, the YI IoT application on the phone stops seeing the camera, is it normal?

cjj25, 2021-08-18:
@fmac83, that's correct. It completely disables the Yi services to prevent any updates being downloaded to the camera.

closed 2021-08-18 by cjj25