code-explorer1234/Yi-RTS3903N-RTSPServer #5: Firmware and success report for Goowls Outdoor Dome camera

nneul, 2021-08-23:
firmware is as backed up with only change being an in place binary edit of wifi SSID and key to equal length strings.

ssid: WIFISSIDABCDE
key: AABBCCDDEEFFGG

https://drive.google.com/file/d/1VcaTG3aUzQCIeBgyfahWlx3CzM-N2yG8/view?usp=sharing

This is firmware 7.1.00.19A_201910181012

Couldn't find listing on amazon any more, but an equivalent link for picture:   https://www.ebay.com/itm/333980231353

nneul, 2021-08-23:
No pink tint noticed on the camera stream.

nneul, 2021-08-23:
Also for a victure sc210 - matches firmware 7.1.00.25A_202002271051 you already have listed, so might not be beneficial to you:

https://drive.google.com/file/d/1BSLPdevCs5wBbKudRmcuSl5rt42gUN6e/view?usp=sharing

nneul, 2021-08-23:
Is it expected that once applied the cameras would stop talking to the cloud services? Saw that with both of them.

cjj25, 2021-08-23:
Hi @nneul, thanks for the feedback! I'll get it added to the Wiki. It completely disables the Yi-Cloud services, if you telnet in and take a look at /home/app/init.sh you can see all the Yi related stuff.

Unfortunately you can't run both services as they take ownership of the video buffer. 

I've also got a working audio RTSPServer, just need to fix some bugs.

Strange request - do you have any internal pictures of the board? I'm currently drawing up a RTS3909N pinout diagram.

nneul, 2021-08-23:
Board pic - not at the moment, but - have had this camera open in the past, so could easily get you some... Will try to do that this evening if I have a few minutes.

cjj25, 2021-08-23:
It looks like your camera (possibly both) could be based on the Tuya [TA-R9820-K2](https://expo.tuya.com/product/322496)/[TA-R9820-G1](https://nullrefer.com/?https://expo.tuya.com/product/459042) (FCC  [XYTA-R9820-K2](https://fcc.report/FCC-ID/2APD7-R9820)) product that's been rebranded.

@nneul, that's great. It'll be useful to have some high res shots. Re: Yi Cloud, are there any specific features you need? Ideally I want to strip Yi-Cloud completely (overwrite the flash) with the key functionality replaced.

I'm yet to obtain a dump of the Tuya based firmware for RTS3903N cameras (anyone reading this?).

Pictures: [External](https://fcc.report/FCC-ID/2APD7-R9820/4214327) and [Internal](https://fcc.report/FCC-ID/2APD7-R9820/4214328)

The [FCC cover letter](https://fcc.report/FCC-ID/2APD7-R9820/4214326) has: 
The product model **XY/TA-R9820-K2 is the same PCB layout**, electronic schematic and Antenna with
* XY/TA-R9820-F9
* XY/TA-R9820-F10
* XY/TA-R9820-K1
* XY/TA-R9820-K3
* XY/TA-R9820-K4
* XY/TA-R9820-K5
* XY/TA-R9820-K6
* XY/TA-R9320-S2
* XY/TA-R9320-S3
* XY/TA-R9420-X2
* XY/TA-R9420-X3
* XY/**TA-R9820-G1**
* XY/TA-R9820-G2
* XY/TA-R9520-V3
* XY/TA-R9520-V6
* TA-R9520-HJ
* TA-FD-BL3
* BW-SHC1
* DG-K2

nneul, 2021-08-23:
Yi-cloud, no specific needs other than I've been doing both random mobile viewing + testing with rtsp. Ideally at some point would just get zoneminder or similar capturing all of these random devices to one place.

nneul, 2021-08-23:
Re pictures, just realized that I didn't have this fully opened, only had it to where the motor drive board was exposed, but not the board within the camera orb itself that has the MCU on it. Will have to try to find some time to see if I can easily get it fully opened.

nneul, 2021-08-28:
[image]
[image]
[image]
[image]
[image]
[image]

hope those are useful to you.

cjj25, 2021-08-29:
These are fantastic! Thank you for taking the time to upload them!

Ammarpic, 2022-07-12:
I have a problem with the camera
I need the basic software
Please help 
camera goowls

Ammarpic, 2022-07-12:
The link is not working, I need the firmware, please

Ammarpic, 2022-07-12:
> firmware is as backed up with only change being an in place binary edit of wifi SSID and key to equal length strings.
>
> ssid: WIFISSIDABCDE key: AABBCCDDEEFFGG [...]

The link is not working, I need the firmware, please