code-explorer1234/Yi-RTS3903N-RTSPServer #6: RTSP does not work (YI IoT Outdoor)

tijo82, 2021-08-24:
I tried your instructions on an YI IoT Outdoor camera but the RTSP does not seem to work. Access via telnet is possible.

VLC says: VLC is unable to open the MRL 'rtsp://192.168.178.42/ch-1_0.h264'.

This is what the log.txt says (I just removed my wifi passcode):

10-24 10:24:07.949 dispatch.c:4480|main() sysversion:7.1.00.19A_201910181012
10-24 10:24:08.216 dispatch.c:727|get_productinfo() 
10-24 10:24:08.216 dispatch.c:731|get_productinfo() product info partition exist!
10-24 10:24:08.216 dispatch.c:735|get_productinfo() product info partition is not empty, read it!
10-24 10:24:08.216 dispatch.c:787|get_productinfo() hardware version is not empty, read it!
10-24 10:24:08.216 dispatch.c:789|get_productinfo()  ptz_en[2] ptz_dir hor[1] ver[1] cds[3] mirror[1] resolution[1] def_x[00] def_y[0] isp[0] 
10-24 10:24:08.217 cloud.c:5399|sys_init() open share mem ok
10-24 10:24:08.555 dispatch.c:1209|get_config() got sn(TeMQbVPzeQ1KBQZs6x2ZkAvoxgceDPrX)
10-24 10:24:08.555 dispatch.c:1210|get_config() got pwd()
10-24 10:24:08.555 dispatch.c:1211|get_config() got ssid()
10-24 10:24:08.555 dispatch.c:1212|get_config() got tnp_init_string(MMFBJPLJICFGKJHFOJFFPMELHPCFFAGCHEEJKPIDINBKDABIPMIIAONIPBCLBBKNPLKPDKPENIDOAJCP)
10-24 10:24:08.555 dispatch.c:4574|main() hw_type(2)
10-24 10:24:08.666 p2p_tnp.c:6741|main() fshare_open ok
10-24 10:24:09.060 p2p_tnp.c:2055|p2p_set_tnp_init_status() p2p_set_tnp_init_status 1 send_msg ok!
10-24 10:24:09.061 dispatch.c:501|choose_server() in choose_server, region_id = 1, api_server = https://plt-api.xiaoyi.com, sname = RealPJ, dlproto = micn
10-24 10:24:09.061 dispatch.c:501|choose_server() in choose_server, region_id = 17, api_server = https://plt-api-us.xiaoyi.com, sname = RealPJ, dlproto = mius
10-24 10:24:09.061 dispatch.c:4731|main() init ok, cost time(972 ms)
10-24 10:24:09.061 mp4record.c:1293|main() recordevent_init finish
10-24 10:24:09.247 dispatch.c:2532|do_mq_process() dispatch got (0x7f)
10-24 10:24:09.247 dispatch.c:4368|do_mq_process() Process (0x7f) end
10-24 10:24:09.566 watch_process.c:116|get_watch_info() check_interval=10
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=dispatch;cmd=cd /home/app;./dispatch &;
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=cloud;cmd=cd /home/app;./cloud &;
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=rmm;cmd=reboot;
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=mp4record;cmd=cd /home/app;./mp4record &;
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=arp_test;cmd=cd /home/app;./arp_test &;
10-24 10:24:09.758 watch_process.c:134|get_watch_info() process=oss;cmd=cd /home/app;./oss &;
10-15 12:23:14.100 dispatch.c:2316|p_worker() DISPATCH_SET_DEFAULT_TIME 1602764594
10-15 12:23:14.163 dispatch.c:1718|do_monitor_wifi() wifi disconnected, now reconnect wifi
10-15 12:23:14.163 dispatch.c:1630|reset_wifi_module() now reset wifi module 
10-15 12:23:14.212 mp4record.c:1304|main() init_finish(1), start_with_reset(0)
10-15 12:23:14.212 mp4record.c:1306|main() mp4 mode start init_finish
10-15 12:23:14.225 mp4record.c:1312|main() fshare_open ok
10-15 12:23:21.405 rmm.c:1401|msg_proc() pid[852]
10-15 12:23:22.483 dispatch.c:1730|do_monitor_wifi() wpa connected , now dhcp
10-24 10:24:08.082 dispatch.c:4480|main() sysversion:7.1.00.19A_201910181012
10-24 10:24:08.362 dispatch.c:727|get_productinfo() 
10-24 10:24:08.363 dispatch.c:731|get_productinfo() product info partition exist!
10-24 10:24:08.363 dispatch.c:735|get_productinfo() product info partition is not empty, read it!
10-24 10:24:08.363 dispatch.c:787|get_productinfo() hardware version is not empty, read it!
10-24 10:24:08.363 dispatch.c:789|get_productinfo()  ptz_en[2] ptz_dir hor[1] ver[1] cds[3] mirror[1] resolution[1] def_x[00] def_y[0] isp[0] 
10-24 10:24:08.363 dispatch.c:1209|get_config() got sn(TeMQbVPzeQ1KBQZs6x2ZkAvoxgceDPrX)
10-24 10:24:08.363 dispatch.c:1210|get_config() got pwd()
10-24 10:24:08.363 dispatch.c:1211|get_config() got ssid()
10-24 10:24:08.363 dispatch.c:1212|get_config() got tnp_init_string(MMFBJPLJICFGKJHFOJFFPMELHPCFFAGCHEEJKPIDINBKDABIPMIIAONIPBCLBBKNPLKPDKPENIDOAJCP)
10-24 10:24:08.363 cloud.c:5399|sys_init() open share mem ok
10-24 10:24:08.584 dispatch.c:4574|main() hw_type(2)
10-24 10:24:08.584 p2p_tnp.c:6741|main() fshare_open ok
10-24 10:24:08.779 p2p_tnp.c:2055|p2p_set_tnp_init_status() p2p_set_tnp_init_status 1 send_msg ok!
10-24 10:24:08.969 dispatch.c:501|choose_server() in choose_server, region_id = 1, api_server = https://plt-api.xiaoyi.com, sname = RealPJ, dlproto = micn
10-24 10:24:08.969 dispatch.c:501|choose_server() in choose_server, region_id = 17, api_server = https://plt-api-us.xiaoyi.com, sname = RealPJ, dlproto = mius
10-24 10:24:08.970 dispatch.c:4731|main() init ok, cost time(887 ms)
10-24 10:24:09.015 mp4record.c:1293|main() recordevent_init finish
10-24 10:24:09.311 dispatch.c:2532|do_mq_process() dispatch got (0x7f)
10-24 10:24:09.312 dispatch.c:4368|do_mq_process() Process (0x7f) end
10-24 10:24:09.421 watch_process.c:116|get_watch_info() check_interval=10
10-24 10:24:09.607 watch_process.c:134|get_watch_info() process=dispatch;cmd=cd /home/app;./dispatch &;
10-24 10:24:09.607 watch_process.c:134|get_watch_info() process=cloud;cmd=cd /home/app;./cloud &;
10-24 10:24:09.608 watch_process.c:134|get_watch_info() process=rmm;cmd=reboot;
10-24 10:24:09.608 watch_process.c:134|get_watch_info() process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
10-24 10:24:09.608 watch_process.c:134|get_watch_info() process=mp4record;cmd=cd /home/app;./mp4record &;
10-24 10:24:09.608 watch_process.c:134|get_watch_info() process=arp_test;cmd=cd /home/app;./arp_test &;
10-24 10:24:09.608 watch_process.c:134|get_watch_info() process=oss;cmd=cd /home/app;./oss &;
10-15 12:23:14.100 dispatch.c:2316|p_worker() DISPATCH_SET_DEFAULT_TIME 1602764594
10-15 12:23:14.116 mp4record.c:1304|main() init_finish(1), start_with_reset(0)
10-15 12:23:14.116 mp4record.c:1306|main() mp4 mode start init_finish
10-15 12:23:14.117 mp4record.c:1312|main() fshare_open ok
10-15 12:23:14.165 dispatch.c:1718|do_monitor_wifi() wifi disconnected, now reconnect wifi
10-15 12:23:14.165 dispatch.c:1630|reset_wifi_module() now reset wifi module 
10-15 12:23:21.006 rmm.c:1401|msg_proc() pid[834]
10-15 12:23:23.540 watch_process.c:171|check_watch_info() cloud crashed!
10-15 12:23:23.540 watch_process.c:171|check_watch_info() p2p_tnp crashed!
10-15 12:23:23.540 watch_process.c:171|check_watch_info() arp_test crashed!
10-15 12:23:24.545 dispatch.c:1730|do_monitor_wifi() wpa connected , now dhcp

cjj25, 2021-08-24:
What's the exact model of the camera?

Could you login via telnet and copy/paste the output of the following

`ps` and `dmesg`

tijo82, 2021-08-25:
This is what the label of the camera says: XY-R9520-V3

This is what ps says:

~ # ps
  PID USER       VSZ STAT COMMAND
    1 root      1428 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW<  [khelper]
   11 root         0 SW<  [writeback]
   12 root         0 SW<  [bioset]
   13 root         0 SW<  [kblockd]
   14 root         0 SW   [khubd]
   15 root         0 SW   [kworker/0:1]
   16 root         0 SW<  [cfg80211]
   17 root         0 SW   [kswapd0]
   18 root         0 SW   [fsnotify_mark]
   19 root         0 SW<  [crypto]
   28 root         0 DW   [enable_swp_task]
   47 root         0 SW<  [deferwq]
   48 root         0 SW   [kworker/u2:1]
   49 root         0 SW<  [kworker/0:1H]
  230 root      1428 S    syslogd
  232 root      1420 S    klogd
  236 root         0 SWN  [jffs2_gcd_mtd5]
  239 root      1428 S    init
  315 root         0 SW   [kworker/u2:2]
  393 root         0 SW   [kworker/0:2]
  402 root         0 SW   [mmcqd/0]
  690 root         0 SW   [RTW_CMD_THREAD]
  795 root      1288 S    wpa_supplicant -c/tmp/wpa_supplicant.conf -g/tmp/wpa
  856 root      1428 S    {exe} ash /var/tmp/sd/wifi/fork_process.sh
  865 root      1428 S    busybox telnetd
  885 root      1428 S    udhcpc -i wlan0 -b -s /home/app/script/default.scrip
  889 root     31864 S <  ./stream
  890 root      2676 S    ./rRTSPServer -d
  901 root      1432 S    -sh
  902 root      1428 R    ps

This is what dmesg says:

~ # dmesg
ip_tables: (C) 2000-2006 Netfilter Core Team
TCP: cubic registered
NET: Registered protocol family 17
Key type dns_resolver registered
mtd1 name is boot
hconf init failed
ALSA device list:
  No soundcards found.
VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
Freeing unused kernel memory: 176K (803e4000 - 80410000)
usb 1-1: new high-speed USB device number 2 using ehci-platform
usb 1-1: New USB device found, idVendor=0bda, idProduct=f179
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: 802.11n
usb 1-1: Manufacturer: Realtek
usb 1-1: SerialNumber: 00e04c000001
jffs2: notice: (235) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
rlx snd internal codec init
soc-audio soc-audio.0.auto: ASoC: machine RLX_INTERN_CARD should use snd_soc_register_card()
soc-audio soc-audio.0.auto:  rlx-codec-digital <-> pcm-platform mapping ok
soc-audio soc-audio.0.auto:  rlx-codec-analog <-> pcm-platform mapping ok
usbcore: registered new interface driver rtl8188fu
rtscam:isp resvd mem addr : 0x00520000, size : 0x1900000
rtscam:rtscam_mem_init v:0xa0520000 p:0x00520000 s:0x00001900
rtscam:rtscam_lock_init
rtscam:rtscam_soc_probe
rtscam:rtscam_hx280_probe
rtscam:hx280enc:HW at base <0x18060000> with ID <0x48317011>
rtscam:rtscam_jpgenc_probe
rtscam:rtscam_osd2_probe
rtscam:rtstream_init
mmc0: new ultra high speed SDR50 SDHC card at address 59b4
mmcblk0: mmc0:59b4 USDU1 14.8 GiB
 mmcblk0: p1
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
hw  =[21131113100000000000000000000000]
gpio=[11111111110011000000000000000000]
[CPLD_PERIPH] CPLD_PERIPH module inited
Started watchdog timer
Started watchdog timer

[PID_LIST] pid_list_init ok, [ ver=May 27 2019, 16:21:04 ]
rtscam:begin to load fw from isp.fw
rtscam:Load firmware size : 131024.
rtscam:Found ISP 1.011 device
rtscam:video device <rts3903-isp> registered
rtscam:rts3903-isp initialized
rtscam:begin to load fw from isp.fw
rtscam:Load firmware size : 131024.
rtscam:begin to load fw from /home/lib/sc2230/isp_jin.fw
rtscam:Load firmware size : 131024.
rtscam:Found ISP 1.011 device
rtscam:video device <rts3903-isp> registered
rtscam:rts3903-isp initialized
usbcore: deregistering interface driver rtl8188fu
usb 1-1: reset high-speed USB device number 2 using ehci-platform
usbcore: registered new interface driver rtl8188fu
Stopped watchdog timer

cjj25, 2021-08-25:
That's a little strange as we can see the `stream` and `rRTSPServer` are running the background so the payload was a success. 

A similar camera was reported to be [working here](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/5#issuecomment-903747367).

Could you provide an upload of the backup directory (firmware dump) - _for full transparency this will have your WiFi password embedded in the mtdblock7 (or 6 .. can't check right now)_

cjj25, 2021-08-25:
I'm also aware of a little bug that could be creating a problem for you. If we try to connect to the RTSPServer too quickly when turning on the camera, you won't get a feed. Have you tried waiting a little longer before connecting?

tijo82, 2021-08-30:
Sorry for the delay - with the new server address I can get a connection. Thanks for debugging!

closed 2021-08-30 by cjj25