code-explorer1234/Yi-RTS3903N-RTSPServer #8: working on victure SC210

fkoteam, 2021-09-01:
Thank you, it's working!
Here is my backup:

https://fromsmash.com/o9WjhZG4zM-ct

My Firmware Version: 7.1.00.19A_201910181012

Only few comments / requests:
-when this hack is working, camera app (YI IoT) doesn't work.
-it would be great (greater) supporting ddns (dyndns) to connect from internet.

Thank you again!

cjj25, 2021-09-02:
Fantastic news! I noticed you were interested in the 'iNite Night Vision Camera' binaries on Hackaday.

If you're still interested in this, I do have a firmware dump.

I've managed to get the official Realtek SDK 'turnkey' version to build an image for one of my cameras with the full Realtek stock experience (RTSP, web interface).

The only issue is each hardware variation differs slightly, for example the U-Boot config needs to be mapped correctly for the smaller / larger size flash chips. It's handy to completely desolder the SPI flash and have an external ZIF adapter, hook up UART and keep testing.

While I'm happy I could get the full Realtek stock experience built, the potential legal copyright issues are stopping me as the binaries are precompiled. I've now switched my attention back to the original Yi firmware and I've spent time reverse engineering the circular buffer of the fshare_frame_buf.

Cross-post from: https://github.com/cjj25/yi-hack-telnet-root/issues/1

fkoteam, 2021-09-02:
I understand your concerns. I wanted inite binaries to try to execute directly on my cam, but your work is much better.
Great work!

closed 2021-09-02 by cjj25

omarelanis, 2024-07-30:
@fkoteam I know this is an old thread, so a long shot but do you remember how you got this working on the Victure 210?

The camera does it's initial checks then goes into the annoying continues 2 beeps every few seconds whilst waiting to be setup using the app.

Can't figure out why it's not doing anything from the SD card, like it's just ignoring it all