code-explorer1234/Yi-RTS3903N-RTSPServer #9: Live picture via rtps to my Lilliput-001 go, comma is working :)

PatrickR68, 2021-09-27:
Hello cjj25, 

many thanks, it works. 
Great.

I have access via rtsp and an app on my Android on the live picture.
Camera looks similar to Dome 1080P

Lilliput-001 go, comma (bought in China via gear....)
Firmware: 7.0.00.73A_201812031453

1) Do you think that "Control PTZ (pan/tilt)" will be available ? Does an schedule exist ?

2) Does this hack blocks home calls to china ?

Many thanks,
Patrick

cjj25, 2021-09-28:
Hi Patrick,

Thanks for the feedback, I've not seen this camera before! 
Once the payload is active then all Yi services (communicating back to China) are killed. 

There is a small chance on first startup where we take advantage of the Yi boot sequence where it could call home. 

The project is currently on hold until I find some more time to continue.

I have successfully managed to get PTZ working on a local camera as a sample but it won't work for all cameras as each camera has it's own GPIO pin map, the signature for this combination is stored in the flash. To make a universal fix, I need to reverse engineer the exact behaviour of reading the signature from flash.

If you could provide a copy of your flash dump, that'll help with reversing the signature of the GPIO pins. However, to be completely transparent.. this flash dump will contain your WiFi password. I recommend sharing a link that expires after download or after a couple of days :)

PatrickR68, 2021-09-28:
Hello cjj25, 

thanks a lot for your answer. 

How can I make the flash dump?
And thanks for the hint relating the WiFi password.

I uploaded some pictures which I took from the open camera. 
But yesterday I closed the cammera  :)

Let me know if I can help you.
Patrick

[go,comma.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7244348/go.comma.zip)

cjj25, 2021-09-28:
Thanks for providing the pictures, they'll come in handy!  It turns out I have the exact same PCB in one of my cameras here.

The payload should have automatically created a backup folder on the root of the SD card for you.

PatrickR68, 2021-09-28:
Your welcome. 
Relating the FW for my one please see the link below and the attached firmware-pack :)

If such links are forbidden, please inform me so that I will delete them.
https://www.gearbest.com/blog/download/gocomma-lilliput-001-ip-camera-firmware-pack-3389
 
Flash dump will come also soon.

Patrick

[2785604 gocomma wireless security IP camera upgrade firmware pack and guide.zip](https://github.com/cjj25/Yi-RTS3903N-
RTSPServer/files/7246003/2785604.gocomma.wireless.security.IP.camera.upgrade.firmware.pack.and.guide.zip)

PatrickR68, 2021-09-28:
Hello cjj25, 

to change the WiFi connection I deleted the camera in the app. 

Now I don't get any connection with the app. 

Attached the log-files with a SD card and your SW. 

Maybe you can see what could be wrong. 

Thanks a lot.

Patrick
[log.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7247623/log.txt)
[log_oss.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7247624/log_oss.txt)

cjj25, 2021-09-28:
Does it connect to WiFi and all you telnet access?

PatrickR68, 2021-09-28:
It starts with start sound, then I can hear the two beep sound for "waiting to connect". 
No WiFi connection. 
I don't know how to use telnet in this situation.

cjj25, 2021-09-28:
Ok, it looks like the removing it from the app has unpaired your device and it's now in pairing mode.

Try use the WiFi trick from the Readme. Failing that, you'll have to pair it again to bypass.

PatrickR68, 2021-09-28:
You mean to use the "wpa_supplicant.conf"
I replaced ssid="SSID_NAME_OF_WIFI" and psk="WIFI_SECRET_KEY" with correct values. 
For WPA it is enough, and the values are in the "", correct?

cjj25, 2021-10-10:
Did you manage to resolve this problem?

PatrickR68, 2021-10-10:
No, I hadn't time to solve it. 
Also I thought maybe to use for my current use a Yi Home 1080p.

closed 2021-12-28 by cjj25