code-explorer1234/Yi-RTS3903N-RTSPServer #15: Resetting the Yi IoT Camera?

Uiasdnmb, 2021-12-26:
Not sure how appropriate this is but this is the only place on the entire internet where anyone discusses this device.

My camera is on 7.1.00.19A_201910181012 fw and it appears to be soft bricked - doesn't react to reset key, when it boots there's no audio cue and never shows up in Yi IoT app.

It does connect to the wifi (can ping it no problem), it toggles IR light and it does print to the log file as well.

<details>
<summary>Click to see log file</summary>

```log
10-24 10:24:08.077 dispatch.c:4480|main() sysversion:7.1.00.19A_201910181012
10-24 10:24:08.329 dispatch.c:727|get_productinfo() 
10-24 10:24:08.329 dispatch.c:731|get_productinfo() product info partition exist!
10-24 10:24:08.329 dispatch.c:755|get_productinfo() product info partition is empty, check if backup exist!
10-24 10:24:08.329 dispatch.c:780|get_productinfo() backup not exist, read tmp files!
10-24 10:24:08.329 dispatch.c:383|get_tmp_did() did not exist, read tmp files
10-24 10:24:08.329 cloud.c:5399|sys_init() open share mem ok
10-24 10:24:08.650 dispatch.c:1209|get_config() got sn(UloZ8hO2..._REDACTED_)
10-24 10:24:08.650 dispatch.c:1210|get_config() got pwd(_REDACTED_)
10-24 10:24:08.650 dispatch.c:1211|get_config() got ssid(_REDACTED_)
10-24 10:24:08.650 dispatch.c:1212|get_config() got tnp_init_string(MMFBJPLJICFGKJHFOJFFPMELHPCFFAGCHEEJKPIDINBKDABIPMIIAONIPBCLBBKNPLKPDKPENIDOAJCP)
10-24 10:24:08.731 dispatch.c:4574|main() hw_type(2)
10-24 10:24:08.731 dispatch.c:4583|main() open_cpld fail
10-24 10:24:08.731 dispatch.c:4608|main() ioctl(536899605) error(25),Inappropriate ioctl for device
10-24 10:24:08.731 dispatch.c:4609|main() ioctl(536899606) error(25),Inappropriate ioctl for device
10-24 10:24:08.731 dispatch.c:4610|main() ioctl(536899603) error(25),Inappropriate ioctl for device
10-24 10:24:08.731 dispatch.c:4637|main() open_ptz fail
10-24 10:24:08.733 dispatch.c:501|choose_server() in choose_server, region_id = 1, api_server = https://plt-api.xiaoyi.com, sname = RealPJ, dlproto = micn
10-24 10:24:08.733 dispatch.c:501|choose_server() in choose_server, region_id = 16, api_server = https://plt-api-de.xiaoyi.com, sname = RealPJ, dlproto = mieu
10-24 10:24:08.733 dispatch.c:4683|main() ioctl(536899612) error(25),Inappropriate ioctl for device
10-24 10:24:08.734 dispatch.c:4731|main() init ok, cost time(656 ms)
10-24 10:24:08.779 mp4record.c:1293|main() recordevent_init finish
10-24 10:24:09.018 p2p_tnp.c:6741|main() fshare_open ok
10-24 10:24:09.324 dispatch.c:2532|do_mq_process() dispatch got (0x7f)
10-24 10:24:09.324 dispatch.c:4368|do_mq_process() Process (0x7f) end
10-24 10:24:09.324 p2p_tnp.c:2055|p2p_set_tnp_init_status() p2p_set_tnp_init_status 1 send_msg ok!
10-24 10:24:09.505 watch_process.c:116|get_watch_info() check_interval=10
12-12 09:35:46.213 watch_process.c:134|get_watch_info() process=dispatch;cmd=cd /home/app;./dispatch &;
12-12 09:35:46.213 watch_process.c:134|get_watch_info() process=cloud;cmd=cd /home/app;./cloud &;
12-12 09:35:46.214 watch_process.c:134|get_watch_info() process=rmm;cmd=reboot;
12-12 09:35:46.214 watch_process.c:134|get_watch_info() process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
12-12 09:35:46.214 watch_process.c:134|get_watch_info() process=mp4record;cmd=cd /home/app;./mp4record &;
12-12 09:35:46.214 watch_process.c:134|get_watch_info() process=arp_test;cmd=cd /home/app;./arp_test &;
12-12 09:35:46.214 watch_process.c:134|get_watch_info() process=oss;cmd=cd /home/app;./oss &;
12-12 09:35:46.551 dispatch.c:2316|p_worker() DISPATCH_SET_DEFAULT_TIME 1639301746
12-12 09:35:46.551 mp4record.c:1304|main() init_finish(1), start_with_reset(0)
12-12 09:35:46.551 mp4record.c:1312|main() fshare_open ok
12-12 09:35:46.551 dispatch.c:1715|do_monitor_wifi() ioctl(536899588) error(25),Inappropriate ioctl for device
12-12 09:35:46.551 dispatch.c:1716|do_monitor_wifi() ioctl(536899594) error(25),Inappropriate ioctl for device
12-12 09:35:46.551 dispatch.c:1718|do_monitor_wifi() wifi disconnected, now reconnect wifi
12-12 09:35:46.551 dispatch.c:1630|reset_wifi_module() now reset wifi module 
12-12 09:35:53.987 rmm.c:1401|msg_proc() pid[820]
12-12 09:35:54.903 dispatch.c:1730|do_monitor_wifi() wpa connected , now dhcp
12-12 09:35:55.250 dispatch.c:1732|do_monitor_wifi() wpa connected , now dhcp finish
12-12 09:35:55.869 watch_process.c:171|check_watch_info() cloud crashed!
12-12 09:35:55.869 watch_process.c:171|check_watch_info() p2p_tnp crashed!
12-12 09:35:55.869 watch_process.c:171|check_watch_info() arp_test crashed!
12-12 09:35:56.277 dispatch.c:1478|check_wifi_connect() bssid(70:4f:57:d9:ee:a1)
12-12 09:35:56.277 dispatch.c:1500|check_wifi_connect() ip(192.168.0.150)
12-12 09:35:56.289 dispatch.c:1511|check_wifi_connect() gw(192.168.0.1)
12-12 09:35:57.345 dispatch.c:1520|check_wifi_connect() gwmac(24:4B:_REDACTED_)
12-12 09:35:57.361 dispatch.c:1530|check_wifi_connect() mask(255.255.255.0)
12-12 09:35:57.392 dispatch.c:1540|check_wifi_connect() mac(44:EF:BF:4C:00:E2)
12-12 09:35:57.535 dispatch.c:1561|check_wifi_connect() signal_quality(81)
12-12 09:35:57.535 dispatch.c:1755|do_monitor_wifi() ioctl(536899585) error(25),Inappropriate ioctl for device
12-12 09:35:57.536 dispatch.c:1756|do_monitor_wifi() ioctl(536899594) error(25),Inappropriate ioctl for device
12-12 09:35:58.593 dispatch.c:1586|check_wifi_connect() signal_quality(88)
12-12 09:36:05.878 watch_process.c:171|check_watch_info() cloud crashed!
12-12 09:36:05.885 watch_process.c:171|check_watch_info() p2p_tnp crashed!
12-12 09:36:06.010 cloud.c:5399|sys_init() open share mem ok
[... 99 lines cut ...]
12-26 21:23:15.152 cloud.c:2465|webapi_do_login() 
12-26 21:23:16.843 dispatch.c:2532|do_mq_process() dispatch got (0x400e)
12-26 21:23:16.844 dispatch.c:4368|do_mq_process() Process (0x400e) end
12-26 21:23:17.154 motion_detect.c:454|modet_process() small action[0/1605] (small:1)!  msarr[24][40]: 94, cnt:2/3, thresh:65, time:1640553797 sec
12-26 21:23:19.930 cloud.c:2121|need_update() ret = 
12-26 21:23:20.469 mp4record.c:938|send_mp4rcd_heartbeat_msg() msg snd success
12-26 21:23:20.471 dispatch.c:2532|do_mq_process() dispatch got (0xe7)
12-26 21:23:20.471 dispatch.c:4104|do_mq_process() got rcd heartbeat
12-26 21:23:20.471 dispatch.c:4368|do_mq_process() Process (0xe7) end
12-26 21:23:24.951 cloud.c:2121|need_update() ret = 
12-26 21:23:25.183 cloud.c:2465|webapi_do_login() 
12-26 21:23:25.671 motion_detect.c:418|modet_process() IsMotion(1/1719)(small:3)------ y-x[20][8]: ms-cnt(thresh):131(ave:86)-13/48(65-6), msSta(60-0):56-21-27-42-85-90-248 time:1497(1640553805-671524). . discd:1, nt:0
12-26 21:23:25.763 motion_detect.c:284|modet_isBiggerMove() Bigger move. old:0, new:28, thresh:20. ms:0-179 oldtick:0 now:1498
12-26 21:23:25.764 rmm.c:865|motion_proc() ###fmy: got a new motion start pause_motion =0 motion =1 unlock_motion=2098 
12-26 21:23:25.766 rmm.c:876|motion_proc() send_motion_start_msg. old:28, new:28
12-26 21:23:25.768 dispatch.c:2532|do_mq_process() dispatch got (0x7c)
12-26 21:23:25.768 dispatch.c:2687|do_mq_process() g_dispatch_info.mmap_info->motion_type  =0, motion_time =[1640553805]
12-26 21:23:25.769 dispatch.c:4368|do_mq_process() Process (0x7c) end
12-26 21:23:25.772 rmm.c:615|send_motion_start_msg() msg snd success
```
</details>

What's most relevant is I was able to run this script to extract the firmware backup AND run RTSPServer without problems.

Is there anyway to force it (through telnet or by reuploading firmware) back to factory settings so it's possible to pair it with IoT app again?

cjj25, 2021-12-27:
Very interesting! 

How were you able to obtain this log, which steps did you take? I'm assuming you telnet into the device and then try run the shell script manually?

When logging in via telnet (not running RTSPServer) what's the output of `ps aux`

Uiasdnmb, 2021-12-27:
It's regular log.txt that's written onto SD card when booting the device normally (when i insert an empty card aka script is not executed).

I'm not able to telnet to the device without running the script, and when I do ps doesn't seem to accept aux argument (only raw command seems to work?).

<details>
<summary>Telnet -> ps when running script</summary>

```
(none) login: root
 _           _         _
| |         | |       | |
| |     ___ | |__  ___| |_ ___ _ __
| |    / _ \| '_ \/ __| __/ _ \ '__|
| |___| (_) | |_) \__ \ ||  __/ |
|______\___/|_.__/|___/\__\___|_|

~ # ps aux
ps: invalid option -- a
~ # ps
  PID USER       VSZ STAT COMMAND
    1 root      1428 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW<  [khelper]
   11 root         0 SW<  [writeback]
   12 root         0 SW<  [bioset]
   13 root         0 SW<  [kblockd]
   14 root         0 SW   [khubd]
   15 root         0 SW   [kworker/0:1]
   16 root         0 SW<  [cfg80211]
   17 root         0 SW   [kswapd0]
   18 root         0 SW   [fsnotify_mark]
   19 root         0 SW<  [crypto]
   28 root         0 DW   [enable_swp_task]
   47 root         0 SW<  [deferwq]
   48 root         0 SW   [kworker/u2:1]
   49 root         0 SW<  [kworker/0:1H]
  230 root      1428 S    syslogd
  232 root      1420 S    klogd
  236 root         0 SWN  [jffs2_gcd_mtd5]
  239 root      1428 S    init
  315 root         0 SW   [kworker/u2:2]
  394 root         0 SW   [kworker/0:2]
  403 root         0 SW   [mmcqd/0]
  668 root         0 SW   [RTW_CMD_THREAD]
  789 root      1284 S    wpa_supplicant -c/tmp/wpa_supplicant.conf -g/tmp/wpa_supplicant-global -Dwext -iwlan0 -B
  848 root      1428 S    {exe} ash /var/tmp/sd/wifi/fork_process.sh
  857 root      1428 S    busybox telnetd
  875 root      1428 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
  879 root     31864 S <  ./stream
  880 root      2640 S    ./rRTSPServer -d
  891 root      1432 S    -sh
  893 root      1428 R    ps
```

</details>

<details>
<summary>Boot log.txt when running script</summary>

```
10-24 10:24:08.169 dispatch.c:4480|main() sysversion:7.1.00.19A_201910181012
10-24 10:24:08.506 cloud.c:5399|sys_init() open share mem ok
10-24 10:24:08.768 dispatch.c:727|get_productinfo() 
[... 21 lines cut ...]
10-24 10:24:09.144 p2p_tnp.c:2055|p2p_set_tnp_init_status() p2p_set_tnp_init_status 1 send_msg ok!
10-24 10:24:09.144 mp4record.c:1293|main() recordevent_init finish
10-24 10:24:09.569 watch_process.c:116|get_watch_info() check_interval=10
10-24 10:24:09.781 watch_process.c:134|get_watch_info() process=dispatch;cmd=cd /home/app;./dispatch &;
10-24 10:24:09.782 watch_process.c:134|get_watch_info() process=cloud;cmd=cd /home/app;./cloud &;
10-24 10:24:09.782 watch_process.c:134|get_watch_info() process=rmm;cmd=reboot;
10-24 10:24:09.782 watch_process.c:134|get_watch_info() process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
10-24 10:24:09.782 watch_process.c:134|get_watch_info() process=mp4record;cmd=cd /home/app;./mp4record &;
10-24 10:24:09.782 watch_process.c:134|get_watch_info() process=arp_test;cmd=cd /home/app;./arp_test &;
10-24 10:24:09.782 watch_process.c:134|get_watch_info() process=oss;cmd=cd /home/app;./oss &;
12-12 09:35:46.236 dispatch.c:2316|p_worker() DISPATCH_SET_DEFAULT_TIME 1639301746
12-12 09:35:46.237 mp4record.c:1304|main() init_finish(1), start_with_reset(0)
12-12 09:35:46.237 mp4record.c:1306|main() mp4 mode start init_finish
12-12 09:35:46.237 mp4record.c:1312|main() fshare_open ok
12-12 09:35:46.237 dispatch.c:1715|do_monitor_wifi() ioctl(536899588) error(25),Inappropriate ioctl for device
12-12 09:35:46.237 dispatch.c:1716|do_monitor_wifi() ioctl(536899594) error(25),Inappropriate ioctl for device
12-12 09:35:46.237 dispatch.c:1718|do_monitor_wifi() wifi disconnected, now reconnect wifi
12-12 09:35:46.237 dispatch.c:1630|reset_wifi_module() now reset wifi module 
12-12 09:35:53.718 rmm.c:1401|msg_proc() pid[828]
12-12 09:35:55.821 watch_process.c:171|check_watch_info() cloud crashed!
12-12 09:35:55.822 watch_process.c:171|check_watch_info() p2p_tnp crashed!
12-12 09:35:55.822 watch_process.c:171|check_watch_info() arp_test crashed!
12-12 09:35:56.009 dispatch.c:1730|do_monitor_wifi() wpa connected , now dhcp
```
</details>

cjj25, 2021-12-27:
Everything looks as it should in that log, with the exception of maybe it's trying to do an update - can't quite tell.

- What's the exact model of the camera?
- Is the camera a PTZ (pan / tilt type)
- When did it stop functioning, did you try gain RTSP and everything stopped working then or is simply stopped?
- Could you provide a firmware dump?

Using the 'altered script', can you login and give me the output of `ls /var/tmp/sd` .. there are numerous checks in the YI binaries for certain files that will prevent from going into full production mode.

Uiasdnmb, 2021-12-27:
It's Yi IoT Outdoor, also known as XY-R9520-V3. It's a static outdoor cam, no tilt or pan. It just stopped functioning randomly one day after a blackout I think.

There's nothing interesting in ls since only content of sd card is `/wifi/config.sh` to act as payload to turn on telnet, other things there get generated after boot.

<details>
<summary>ls command here</summary>

```
/var/tmp/sd # ls -l
drwxr-xr-x    2 root     root         32768 Dec 26 19:57 System Volume Information
-rwxr-xr-x    1 root     root       2506752 Dec 27 23:05 core.mp4record.10576
-rwxr-xr-x    1 root     root       2506752 Dec 27 23:05 core.mp4record.10624
-rwxr-xr-x    1 root     root       4784128 Dec 27 23:05 core.mp4record.10673
-rwxr-xr-x    1 root     root       2506752 Dec 27 23:05 core.mp4record.475
-rwxr-xr-x    1 root     root       2351104 Dec 27 14:21 core.mp4record.477
-rwxr-xr-x    1 root     root        215837 Dec 27 23:14 log.txt
-rwxr-xr-x    1 root     root           528 Dec 27 22:28 log_oss.txt
drwxr-xr-x    2 root     root         32768 Dec 27 23:05 record
drwxr-xr-x    2 root     root         32768 Aug  9 14:47 wifi
/var/tmp/sd # ls -l wifi
-rwxr-xr-x    1 root     root           998 Dec 27 15:16 config.sh
```
</details>

I was also browsing around the filesystem but it tells me nothing interesting.
Just a wild shot - since there's no audio cue after boot I tried to run `amixer` command but it only errors out with "amixer: can't load library ;libasound.so.2'". Should this even work in the first place?

I'll update this when I upload fw dump.

cjj25, 2021-12-28:
Thanks for sharing the firmware dump. I'll take a look tomorrow. 

However, there is something in your log that might give us a clue as to what's going on.  The blob that contains the product info might be corrupt or missing (notice the get_productinfo method failing in your log). 

To check your audio, see my [previous comment here.](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/12#issuecomment-944213837)

cjj25, 2021-12-28:
I've just looked into your firmware dump and the product info does appear to exist (mtdblock7) with the exception of it being 10 bytes larger than a previously submitted dump matching your firmware.

Here's something for you to try..

Do a fresh boot and telnet in, then do the following:

`cd /var/tmp/sd/
mv wifi wifi_old
cd /home/app/
./init.sh
`
Then post the output here.

cjj25, 2021-12-28:
See a user who posted their logs and device info with the same [firmware here](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/6).

Notice the product information partition check.

mgerald21, 2021-12-28:
I own the same device XY-R9520-V3, but I'm unsure about the firmware version. 
The camera is currently unpaired and
`/wifi/config.sh` 
is not executed.

cjj25, 2021-12-28:
I'm eager to know how you get on with the mtdblock6 partition as it's something I've been planning to reverse engineer for a while (lack of time).

The blob is loaded into `load_cpld_ssp` which loads the device driver YI uses to interact with various aspects of the camera such as IR cut, light sensor (ADC) and pan tilt.  The string of numbers each represent a function and/or GPIO pin map.

Here are some examples

### Goowls Outdoor Dome Camera - [product link ](https://www.amazon.co.uk/Security-Outdoor-Surveillance-Waterproof-Detection/dp/B083J11JG8)
Pan/Tilt = YES
<details><summary>Possible PCB (click to view)</summary>
<p>

Tuya [TA-R9820-K2](https://nullrefer.com/?https://expo.tuya.com/product/322496)/[TA-R9820-G1](https://nullrefer.com/?https://expo.tuya.com/product/459042) (FCC  [XYTA-R9820-K2](https://fcc.report/FCC-ID/2APD7-R9820))

</p>
</details>
```
0x168 : 12232123100000000000000000000000
0x204 : 11111111110011000000000000000000
```

### Victure SC210 - [product link ](https://uk.govicture.com/products/victure-sc210-1080p-hd-baby-monitor-with-camera?variant=40701675569339)
Pan/Tilt = YES
<details><summary>Possible PCB (click to view)</summary>
<p>

XY/TA-R9820-F12, XY/TA-R9820-F11, XY/TA-R9820-F13, XY/TA-R9820-
F14, XY/TA-R9820-K7, XY/TA-R9820-K8, XY/TA-R9820-K9, XY/TA-R9820-K10,
XY/TA-R9820-Q11, IPC-A3610-A111, IPC/TA-A3610-XHR, XY/TA-R9520-V8, XY/TA-
R9520-V9, IPC-R9520-V7, TA-A3410-R6, IPC-C3310-R7 , XY-W102R, TA-FD-BL6,
TA-FD-BL7, TA-R9522-AFD

FCC ID: 2APD7-XYTAR9820F12
https://fccid.io/2APD7-XYTAR9820F12/Internal-Photos/09-77622-IntPho-4381529
</p>
</details>

Firmware: 7.1.00.25A_202002271051
```
0x168 : 11131123100000000000000000000000
0x204 : 11111111110011000000000000000000
```
Firmware: 7.1.00.19A_201910181012
```
0x168 : 11131123100000000000000000000000
0x204 : 11111111110011000000000000000000
```
Note: Identical, firmware is probably interchangeable 

### Yi Outdoor Camera XY-R9520-V3
Pan/Tilt = NO
Possible PCB : Same as Victure SC210?
Firmware: 7.1.00.19A_201910181012
```
0x168 : 21131113100000000000000000000000
0x204 : 11111111110011000000000000000000
```
Notes compared to SC210 (which is pan tilt)
0x204 has the same layout - I believe this to be the GPIO map
0x168 differs - I believe this to be the product features map (so the software knows it has pan/tilt etc)

Uiasdnmb, 2021-12-28:
I ran `init.sh` after removing the payload but it didn't do anything. It was actually stuck in an infinite loop of trying to update and failing, it never completed and kept running.
After 4 minutes which I think is the recording segment cutoff telnet got polluted with `12-28 14:15:44.165 mp4record.c:1063|record_file() mp4 file create failed` that appeared every 0.04 second so I shut down the device at this point.
It didn't write anything to the log, everything was output in telnet window but it was full of invalid calls to `/backup/cloudAPI` that spit out their entire help manual and list of commands.

Copying mtdblock6 seemed to have helped - device boots with audio cue and pressing reset button does restart it into pairing mode.
I can run the app and show it the QR code and it connects to wifi.
After that though... I'm not sure if it's the apps fault or something is still wrong with the camera because pairing always times out and it doesn't show upp in camera list.

<details>
<summary></summary>

```
10-24 10:24:08.819 dispatch.c:4480|main() sysversion:7.1.00.19A_201910181012
10-24 10:24:09.069 dispatch.c:727|get_productinfo() 
10-24 10:24:09.069 dispatch.c:731|get_productinfo() product info partition exist!
10-24 10:24:09.069 dispatch.c:735|get_productinfo() product info partition is not empty, read it!
10-24 10:24:09.069 dispatch.c:787|get_productinfo() hardware version is not empty, read it!
10-24 10:24:09.069 dispatch.c:789|get_productinfo()  ptz_en[2] ptz_dir hor[1] ver[1] cds[3] mirror[1] resolution[1] def_x[00] def_y[0] isp[0] 
10-24 10:24:09.070 dispatch.c:1209|get_config() got sn(Yg43nSTZ2pd43Y2DZ8YB1NqhsgYr4frS)
10-24 10:24:09.070 dispatch.c:1210|get_config() got pwd()
10-24 10:24:09.071 dispatch.c:1211|get_config() got ssid()
10-24 10:24:09.071 dispatch.c:1212|get_config() got tnp_init_string()
10-24 10:24:09.071 dispatch.c:1220|get_config() reset because no wifi config
10-24 10:24:09.071 dispatch.c:374|send_saveconfig_msg() send_saveconfig_msg msg snd success
10-24 10:24:09.141 dispatch.c:4574|main() hw_type(2)
10-24 10:24:09.563 dispatch.c:501|choose_server() in choose_server, region_id = 1, api_server = https://plt-api.xiaoyi.com, sname = RealPJ, dlproto = micn
10-24 10:24:09.563 dispatch.c:501|choose_server() in choose_server, region_id = 16, api_server = https://plt-api-de.xiaoyi.com, sname = RealPJ, dlproto = mieu
10-24 10:24:09.564 dispatch.c:4731|main() init ok, cost time(799 ms)
10-24 10:24:09.571 dispatch.c:962|save_config() got pwd()()
10-24 10:24:09.572 dispatch.c:963|save_config() got ssid()()
10-24 10:24:09.572 dispatch.c:964|save_config() config save finish 1
10-24 10:24:09.625 dispatch.c:2532|do_mq_process() dispatch got (0x7f)
10-24 10:24:09.625 dispatch.c:4368|do_mq_process() Process (0x7f) end
10-24 10:24:09.870 dispatch.c:2236|p_worker() save_config
04-19 11:54:57.100 dispatch.c:2316|p_worker() DISPATCH_SET_DEFAULT_TIME 1555674897
04-19 11:54:57.564 watch_process.c:116|get_watch_info() check_interval=10
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=dispatch;cmd=cd /home/app;./dispatch &;
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=cloud;cmd=cd /home/app;./cloud &;
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=rmm;cmd=reboot;
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=mp4record;cmd=cd /home/app;./mp4record &;
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=arp_test;cmd=cd /home/app;./arp_test &;
04-19 11:54:57.677 watch_process.c:134|get_watch_info() process=oss;cmd=cd /home/app;./oss &;
04-19 11:55:03.921 rmm.c:1401|msg_proc() pid[806]
04-19 11:55:04.127 dispatch.c:2532|do_mq_process() dispatch got (0x1002)
04-19 11:55:04.127 dispatch.c:4364|do_mq_process() invalid msg 0x1002
04-19 11:55:04.127 dispatch.c:4368|do_mq_process() Process (0x1002) end
04-19 11:55:04.127 rmm.c:1441|msg_proc() got RMM_SPEAK_WAIT
04-19 11:55:04.127 rmm.c:1235|send_wait_msg() msg snd success
04-19 11:55:04.314 rmm.c:1750|zbar_proc() Got raw video size(640,360)
04-19 11:55:07.357 watch_process.c:171|check_watch_info() arp_test crashed!
04-19 11:55:12.686 rmm.c:1806|zbar_proc() decoded QR-Code symbol "b=EUmqGJWGvSWKVwgK&s=RFBfS3_REDACTED_==&p=CQt5B_REDACTED_"
04-19 11:55:12.687 rmm.c:1314|trans_info() trans_json info=b=EUmqGJWGvSWKVwgK&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw;key=&p=
04-19 11:55:12.687 rmm.c:1386|trans_info() trans_json result=CQt5BwUpPAFw
04-19 11:55:12.688 rmm.c:1314|trans_info() trans_json info=b=EUmqGJWGvSWKVwgK&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw;key=&s=
04-19 11:55:12.688 rmm.c:1386|trans_info() trans_json result=RFBfS3VjaG5pYQ==
04-19 11:55:12.688 rmm.c:1314|trans_info() trans_json info=b=EUmqGJWGvSWKVwgK&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw;key=b=
04-19 11:55:12.689 rmm.c:1386|trans_info() trans_json result=EUmqGJWGvSWKVwgK
04-19 11:55:12.689 rmm.c:1828|zbar_proc() rmm got /tmp/got_wpa ssid(_REDACTED_) pwd(_REDACTED_) bind(EUmqGJWGvSWKVwgK)
04-19 11:55:12.690 dispatch.c:2532|do_mq_process() dispatch got (0x70)
04-19 11:55:12.690 dispatch.c:501|choose_server() in choose_server, region_id = 16, api_server = https://plt-api-de.xiaoyi.com, sname = RealPJ, dlproto = mieu
04-19 11:55:12.691 rmm.c:1302|send_wifi_conf_msg() msg snd success
04-19 11:55:12.801 dispatch.c:2578|do_mq_process() got pwd(123AVCS98)
04-19 11:55:12.801 dispatch.c:93|send_connectting_msg() msg snd success
[... cut ...]
12-28 14:40:23.695 cloud.c:2315|webapi_do_reset_by_specified_server() cmd = /backup/cloudAPI -c 139 -keySec 8lKlQBD766fw6Djt -url https://plt-api-de.xiaoyi.com/v4/ipc/reset -uid TNPXGAB-523068-MVNLZ -version 7.1.00.19A_201910181012 -mac 44:EF:BF:_REDACTED_
12-28 14:40:23.783 p2p_tnp.c:6368|state_statistics() check_login fail 0
12-28 14:40:23.784 dispatch.c:2532|do_mq_process() dispatch got (0x82)
12-28 14:40:23.784 dispatch.c:4368|do_mq_process() Process (0x82) end
12-28 14:40:24.765 cloud.c:2324|webapi_do_reset_by_specified_server() {"code":"20243"}

```
</details>

cjj25, 2021-12-28:
Ok, well that's great news.

Unfortunately, I think your device is now linked / identifying itself as the camera we originally took that mtdblock6 from and YI has linked it to **THEIR** account.

However, all is not lost. Do you have a QR code on your camera? If so, what's the output / serial when scanned. 

Check the camera for any serial numbers or IDs on the outside - post back.

Uiasdnmb, 2021-12-28:
<details><summary>Output of QR code is</summary>A0010000CB3LB0191106</details>

cjj25, 2021-12-28:
Ok, try flash this version (attached)
[mtdblock6.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7785283/mtdblock6.zip)

cjj25, 2021-12-28:
We can also obtain the other DID value from the sniffing the data to the mobile app (if it's still paired to your account)

Uiasdnmb, 2021-12-28:
Pairing still timing out, I think log is the same aside from updated device id.
Sadly removing and trying to pair it again was one of the first thing I attempted so we won't be able to sniff it

<details>
<summary>Another failed pairing</summary>

```
10-24 10:24:08.493 cloud.c:5399|sys_init() open share mem ok
10-24 10:24:08.728 dispatch.c:4480|main() sysversion:7.1.00.19A_201910181012
10-24 10:24:08.990 dispatch.c:727|get_productinfo() 
10-24 10:24:08.991 dispatch.c:731|get_productinfo() product info partition exist!
10-24 10:24:08.991 dispatch.c:735|get_productinfo() product info partition is not empty, read it!
10-24 10:24:08.991 dispatch.c:787|get_productinfo() hardware version is not empty, read it!
10-24 10:24:08.991 dispatch.c:789|get_productinfo()  ptz_en[2] ptz_dir hor[1] ver[1] cds[3] mirror[1] resolution[1] def_x[00] def_y[0] isp[0] 
10-24 10:24:08.991 dispatch.c:1209|get_config() got sn(ZfdWm7FeGWztsplBDlIzRWLYZU7WOFNn)
10-24 10:24:08.991 dispatch.c:1210|get_config() got pwd()
10-24 10:24:08.991 dispatch.c:1211|get_config() got ssid()
10-24 10:24:08.991 dispatch.c:1212|get_config() got tnp_init_string()
10-24 10:24:08.991 dispatch.c:1220|get_config() reset because no wifi config
10-24 10:24:08.991 dispatch.c:374|send_saveconfig_msg() send_saveconfig_msg msg snd success
10-24 10:24:08.996 dispatch.c:4574|main() hw_type(2)
10-24 10:24:09.400 dispatch.c:501|choose_server() in choose_server, region_id = 1, api_server = https://plt-api.xiaoyi.com, sname = RealPJ, dlproto = micn
10-24 10:24:09.400 dispatch.c:501|choose_server() in choose_server, region_id = 16, api_server = https://plt-api-de.xiaoyi.com, sname = RealPJ, dlproto = mieu
10-24 10:24:09.401 dispatch.c:4731|main() init ok, cost time(742 ms)
10-24 10:24:09.405 dispatch.c:962|save_config() got pwd()()
10-24 10:24:09.410 dispatch.c:963|save_config() got ssid()()
10-24 10:24:09.410 dispatch.c:964|save_config() config save finish 1
10-24 10:24:09.516 dispatch.c:2532|do_mq_process() dispatch got (0x7f)
10-24 10:24:09.516 dispatch.c:4368|do_mq_process() Process (0x7f) end
10-24 10:24:09.620 watch_process.c:116|get_watch_info() check_interval=10
10-24 10:24:09.754 watch_process.c:134|get_watch_info() process=dispatch;cmd=cd /home/app;./dispatch &;
10-24 10:24:09.754 watch_process.c:134|get_watch_info() process=cloud;cmd=cd /home/app;./cloud &;
10-24 10:24:09.754 watch_process.c:134|get_watch_info() process=rmm;cmd=reboot;
10-24 10:24:09.755 watch_process.c:134|get_watch_info() process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
10-24 10:24:09.755 watch_process.c:134|get_watch_info() process=mp4record;cmd=cd /home/app;./mp4record &;
10-24 10:24:09.755 watch_process.c:134|get_watch_info() process=arp_test;cmd=cd /home/app;./arp_test &;
10-24 10:24:09.755 watch_process.c:134|get_watch_info() process=oss;cmd=cd /home/app;./oss &;
10-24 10:24:09.776 dispatch.c:2236|p_worker() save_config
04-19 11:54:57.100 dispatch.c:2316|p_worker() DISPATCH_SET_DEFAULT_TIME 1555674897
04-19 11:55:03.704 rmm.c:1401|msg_proc() pid[798]
04-19 11:55:03.795 dispatch.c:2532|do_mq_process() dispatch got (0x1002)
04-19 11:55:03.795 dispatch.c:4364|do_mq_process() invalid msg 0x1002
04-19 11:55:03.795 dispatch.c:4368|do_mq_process() Process (0x1002) end
04-19 11:55:03.795 rmm.c:1441|msg_proc() got RMM_SPEAK_WAIT
04-19 11:55:03.795 rmm.c:1235|send_wait_msg() msg snd success
04-19 11:55:04.055 rmm.c:1750|zbar_proc() Got raw video size(640,360)
04-19 11:55:06.949 watch_process.c:171|check_watch_info() arp_test crashed!
04-19 11:55:16.045 dispatch.c:2532|do_mq_process() dispatch got (0x1002)
04-19 11:55:16.046 dispatch.c:4364|do_mq_process() invalid msg 0x1002
04-19 11:55:16.046 dispatch.c:4368|do_mq_process() Process (0x1002) end
04-19 11:55:16.047 rmm.c:1441|msg_proc() got RMM_SPEAK_WAIT
04-19 11:55:16.049 rmm.c:1235|send_wait_msg() msg snd success
04-19 11:55:24.983 rmm.c:1806|zbar_proc() decoded QR-Code symbol "b=EUZgxW5_D&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw"
04-19 11:55:24.984 rmm.c:1314|trans_info() trans_json info=b=EUZgxW5sadfjzXJJ&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw;key=&p=
04-19 11:55:24.990 rmm.c:1386|trans_info() trans_json result=CQt5BwUpPAFw
04-19 11:55:24.991 rmm.c:1314|trans_info() trans_json info=b=EUZgxW5sadfjzXJJ&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw;key=&s=
04-19 11:55:24.991 rmm.c:1386|trans_info() trans_json result=RFBfS3VjaG5pYQ==
04-19 11:55:24.992 rmm.c:1314|trans_info() trans_json info=b=EUZgxW5sadfjzXJJ&s=RFBfS3VjaG5pYQ==&p=CQt5BwUpPAFw;key=b=
04-19 11:55:24.992 rmm.c:1386|trans_info() trans_json result=EUZgxW5sadfjzXJJ
04-19 11:55:24.992 rmm.c:1828|zbar_proc() rmm got /tmp/got_wpa ssid(_REDACTED_) pwd(_REDACTED_) bind(EUZgxW5sadfjzXJJ)
04-19 11:55:24.994 dispatch.c:2532|do_mq_process() dispatch got (0x70)
04-19 11:55:24.994 dispatch.c:501|choose_server() in choose_server, region_id = 16, api_server = https://plt-api-de.xiaoyi.com, sname = RealPJ, dlproto = mieu
04-19 11:55:24.994 rmm.c:1302|send_wifi_conf_msg() msg snd success
04-19 11:55:25.092 dispatch.c:2578|do_mq_process() got pwd(_REDACTED_)
04-19 11:55:25.092 dispatch.c:93|send_connectting_msg() msg snd success
04-19 11:55:25.092 dispatch.c:4368|do_mq_process() Process (0x70) end
04-19 11:55:25.092 dispatch.c:2532|do_mq_process() dispatch got (0x1004)
04-19 11:55:25.093 dispatch.c:4364|do_mq_process() invalid msg 0x1004
04-19 11:55:25.094 dispatch.c:4368|do_mq_process() Process (0x1004) end
04-19 11:55:25.738 dispatch.c:1718|do_monitor_wifi() wifi disconnected, now reconnect wifi
[... cut ...]
12-28 16:07:12.921 dispatch.c:4368|do_mq_process() Process (0x82) end
12-28 16:07:21.631 cloud.c:2324|webapi_do_reset_by_specified_server() {"code":"20243"}
12-28 16:07:21.945 dispatch.c:2532|do_mq_process() dispatch got (0x400e)
12-28 16:07:21.945 dispatch.c:4368|do_mq_process() Process (0x400e) end
12-28 16:07:22.923 p2p_tnp.c:6368|state_statistics() check_login fail 0
12-28 16:07:22.923 dispatch.c:2532|do_mq_process() dispatch got (0x82)
12-28 16:07:22.923 dispatch.c:4368|do_mq_process() Process (0x82) end

```
</details>

cjj25, 2021-12-28:
I think there might be something wrong with the Yi services at the moment. 

@mgerald21 is having the exact same problem with a stock (brand new) camera.

I've just tried to pair two separate cameras here and they all timeout too.

I guess it's too early to tell whether the patch to your mtdblock6 has worked?

Uiasdnmb, 2021-12-28:
Yeah I just wanted to report what's happening as soon as possible.
Guess we have to give it a bit of time and I'll try to pair it again later, I was thinking about unpairing one of my working cams to see if it works but if you say you can't pair anything either I won't be doing that.

cjj25, 2021-12-28:
Here's another report (21 hours ago) of similar problem on [Yi Community forum.](https://forum.yitechnology.com/t/trouble-pairing-yi-dome-camera-1080p/1347/167)

cjj25, 2021-12-30:
Just checking to see if you've had any luck yet?

Uiasdnmb, 2021-12-30:
I reset my other camera (Smart IP Camera, XY-R9620-K2) and it works and pair succesfully.

I've compared logs of pairing and the difference seems to be that in working IP camera I get:

```
04-19 11:55:37.912 cloud.c:2303|webapi_do_reset_by_specified_server() now do reset 
04-19 11:55:37.912 cloud.c:2315|webapi_do_reset_by_specified_server() cmd = /backup/cloudAPI -c 139 -keySec kdMxv_REDACTED_ -url https://plt-api-de.xiaoyi.com/v4/ipc/reset -uid TNPXGAB-_REDACTED_ -version 7.1.00.19A_201910181012 -mac E0:09:BF:_REDACTED_
12-30 14:34:22.679 cloud.c:2324|webapi_do_reset_by_specified_server() {"code":"20000"}
12-30 14:34:22.679 cloud.c:2330|webapi_do_reset_by_specified_server() webapi_do_reset success {"code":"20000"}
12-30 14:34:23.610 cloud.c:2426|webapi_do_login() now do login 
12-30 14:34:23.611 cloud.c:2453|webapi_do_login() cmd = /backup/cloudAPI -c 138 -url "https://plt-api-de.xiaoyi.com/v4/ipc/on_line" (...)
12-30 14:34:24.416 cloud.c:2465|webapi_do_login() {"code":"20000", (...) }
12-30 14:34:24.417 cloud.c:2482|webapi_do_login() webapi_do_login success
then it does webapi_do_bindkey() and webapi_get_dev_info()
```

While in broken cam I get `webapi_do_reset_by_specified_server() {"code":"20243"}` about 4 times before it gives up and never goes to login.
I tried to force call `backup/cloudApi -c138` through telnet with mac and keysec from this device but it also returned 20243.

cjj25, 2021-12-30:
Very interesting!

Now you can do the actual pairing, did you try the pairing with the original device blob I sent (mtdblock6) rather than the patched version? 

Perhaps I could give you a mtdblock6 from a device I know to be 100% offline and never paired.

Uiasdnmb, 2021-12-30:
Hope you didn't misunderstand, the broken camera still refuses to pair I just have another similar one lying around that has no issues so I just used to see how logs of successful pairing look like.

cjj25, 2021-12-30:
I've got a feeling it's checking the mac, keysec and _uid/did_ on the server end.

Did you copy the _EXACT_ line from the camera that does pair (**including the uid**) and try run it on the camera that fails?

Uiasdnmb, 2021-12-30:
Tried few combinations, exact line from working camera, line with swapped MAC only, line with swapped keySec, line with same keySec.

Only thing I accomplished is by running reset command with keySec from working cam it got removed from the app even though I used mac of broken one.

It turns out that wasn't the best idea, looks like their server is messed up and enters some kind of corrupted state as now even the camera that worked refuses to pair and times out (also gets code 20243 in logs).

cjj25, 2021-12-30:
I've just done some digging in the firmware and it looks like the reset command does indeed remove act as an 'unpair' type of call.

However, there "on_line" endpoint might be what you really want and in the logs it's truncated.

This is the format for the call: `%s -c 138 -url "%s/v4/ipc/on_line" -key %s -keySec %s -uid %s -version %s -ssid "%s" -mac %s -ip %s -signal_quality %d -packetloss %d -p2pconnect %d -p2pconnect_success %d -apmode %s -tfstat %d`

The log mentioned `cloudAPI` but this is misleading, it's actually `/home/app/cloud`

Here's some other endpoints for you to investigate:
```
%s -c 145 -url "%s/vmanager/ipc/firmware/upgrade/silent" -did %s -sname %s -version %s 
%s -c 139 -keySec %s -url %s/v4/ipc/reset -uid %s -version %s -mac %s 
%s -c 137 -key %s -keySec %s -url %s/v4/ipc/qr_bind -uid %s 
%s -c 138 -url "%s/v4/ipc/on_line" -key %s -keySec %s -uid %s -version %s -ssid "%s" -mac %s -ip %s -signal_quality %d -packetloss %d -p2pconnect %d -p2pconnect_success %d -apmode %s -tfstat %d 
%s -c 311 -url %s/v4/ipc/check_did -uid %s -keySec %s
%s -c 142 -url %s/v4/ipc/deviceinfo -uid %s -keySec %s
%s -c 414 -url %s/v4/ipc/upload_data -uid %s -keySec %s -filename %s
%s -c 192 -url "%s/ipc/monitor/heartbeat" -did %s -version %s -mac %s -heart_time %llu -p2p_status %d 
%s -c 144 -url "%s/ipc/firmware/update" -seq %d -did %s -currentCode %s -targetCode %s -region %s -status %d 
%s -c 136 -url %s/v2/ipc/sync_time 
```

Notice the QR bind? Try this...

1. Find the uid code from either the logs or `strings mtdblock6`
2. Create a QR code on [https://www.qr-code-generator.com/](https://www.qr-code-generator.com/) with the uid
3. Try scan this QR code at the pairing screen on the mobile app
4. Nothing interesting happening? Do the same but with the DID code (again use step one to find it, it starts with A) 
5. Creating a QR code with that DID that I gave you in mtdblock6 might actually get you paired properly...
6. Nothing interesting? Do all the above but sniff the packets being sent from the camera.. When it scans the DID code, it might automatically return the UID from the cloud?

Uiasdnmb, 2021-12-30:
Isn't UID the User ID (aka account that tried to pair w/ device)? Pretty sure that info comes from the QR code and there's no issue with fetching that.
Right now I'm gonna assume it's server fault as I did repeat pairing calls that succeeded before but they just returned error code from server. Maybe I'll try playing with it a bit tomorrow but I hope there's some kind of meaningful feedback from the official forum (I'll link post here when it appears bc apparently they have to approve it).

cjj25, 2021-12-30:
I assumed it was the UID too but the logs appear to give different information..

`12-28 16:06:28.106 cloud.c:2315|webapi_do_reset_by_specified_server() cmd = /backup/cloudAPI -c 139 -keySec 8lKlQBD766fw6Djt -url https://plt-api-de.xiaoyi.com/v4/ipc/reset -uid TNPXGAB-523068-MVNLZ -version 7.1.00.19A_201910181012 -mac 44:EF:BF:_REDACTED_`

Uiasdnmb, 2021-12-30:
I don't seem to be able to feed any arguments into `/home/app/cloud`, it just tries to invoke reset 3 times then says "pairing timed out".
Endpoints don't seem to matter I get 20243 returned from everything aside from sync time and update firmware (and check_did returns 404).
I think the key to it remains running a successful reset but it might be messed up on server side because as I mentioned - I unpaired and made working cam useless by running cloudAPI reset with its keySec and UID on broken one.

cjj25, 2021-12-31:
I'm happy to give this a whirl for you, I have around 5 donor cameras floating around that I used for reverse engineering.

To ensure we don't waste any time / black listing the DID or UID, can you summarise 1. the initial problem and 2. the things you've tried (commands) and then I'll work from there?

I don't believe the 20243 command is a ban (I hope) but more of a status, possibly an unpaired status. 

I'll capture the packets for the pairing processing and see exactly what's expected to see if we can make sense of these statuses a bit more.

Uiasdnmb, 2021-12-31:
I don't know what else can I say, I just try to summarize everything:

- we've "fixed" broken cam (cam1) with patched mtdblock, but it wouldn't go through with pairing
- i put SD card in working cam (cam2) to see how log.txt of paired cam looks
- i unpaired cam2 and successfully re-paired it and stored log.txt of that event
- now i started copying commands called by cam2 in cam1 through telnet. I was replacing UID and keySec with what cam1 had in logs, but that only resulted in 20243.
- in the end I executed reset command (-c 139) exactly as it was in cam2 but on cam1. I'm not 100% sure but I think cam2 was online when I did that. This returned 20000.
- now cam2 disappeared from the app and it's no longer possible to pair either cam

Whether I include payload doesn't seem to matter, but I still have to show the QR code before it's executed and telnet opens up.

cjj25, 2022-01-06:
Not had chance to look into this yet as working on different project. 

Great research you've done though!  The camera with the different status, maybe you can pair this one now?

Uiasdnmb, 2022-01-06:
Cam1 still doesn't pair anywhere. 
I've actually had a bit of regret earlier as I realized I made a mistake (before posting here) - SD card I was using had logs of cam1 from when it was working but I formatted it to play around with scripts and now it's unrecoverable. Now I know it contained keySec and UID that could be used to revive the cam.

cjj25, 2022-01-06:
I feel your pain! 

Did you do a low level format or high level on the SD card? Most formatting tools do high level by default which leaves the original data behind.

Perhaps try a tool like Recuva or GetDataBackNTFS / FAT for Windows or PhotoRec for linux...

It's worth a try?

Uiasdnmb, 2022-01-06:
I did try data recover but as I said it's the card that I was using to push scripts backups etc. onto. Log is gone but I could still see 30gb of old recordings.

cjj25, 2022-01-06:
Give the support team here a shot (the live chat) https://forum.yitechnology.com/t/pairing-has-timed-out-for-all-4x-newly-yi-home-camera-1080p-ai/5832/7

Don't mention any of what we've done... but report back.

I've just tried to pair 4 cameras, all saying "Pairing has timed out"

Uiasdnmb, 2022-01-06:
They don't appear to have any brand printed on them, just your default OEM design I got from Aliexpres.

cjj25, 2022-03-16:
@Uiasdnmb, did you manage to make any progress on this / contact the support?

Uiasdnmb, 2022-03-16:
Nope, I gave up and bought cameras from other brand. Support was of no help and were just deflective since they're third party cams.

I guess we've learned that it's possible to recreate damaged mtdblock6 if you pull the sdcard and get UID and keysec from logs (or from a backup), and as long as you have NOT deleted cam from the app it should work again. Repairing it is just a mess and cams seem to get randomly banned or stuck in pairing limbo.

cjj25, 2022-03-16:
Expensive research but at least we learned something. I use Reolink cameras as my main security cameras but the cheapo ones for nature cameras.

Thanks for the update

closed 2022-03-16 by cjj25