code-explorer1234/Yi-RTS3903N-RTSPServer #16: Yi IoT Camera - XY-R9520-V3 not running config.sh

mgerald21, 2021-12-28:
I own the device XY-R9520-V3 with firmware 7.1.00.19A_201910181012
The camera is currently unpaired and SD card is formatted properly.

`# mkfs.vfat -F 32 /dev/sdb1`

But /wifi/config.sh is not executed. Any suggestions are welcome.

cjj25, 2021-12-28:
I believe the problem you're experiencing is that the device doesn't have any WiFi credentials stored in flash. 

- What's your exact camera model?
- Is it WiFi only does it have an ethernet port?
- Is the camera beeping when you turn it on, indicating it wants pairing?

Here's something for you to try:

- Create a folder on the root of the SD card called 'Factory'
- Edit the [wpa_supplicant_sample.conf](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/7763840399c08659c6b2e875bc4bca5a30f7f58b/build/sd_payload/wpa_supplicant_sample.conf) file with your WiFi credentials and copy it to the `Factory` folder
- Rename the `wpa_supplicant_sample.conf` to `wpa_supplicant.conf` ([see why here](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/7763840399c08659c6b2e875bc4bca5a30f7f58b/build/sd_payload/wifi/fork_process.sh#L19))

Does the camera now connect to the WiFi?

mgerald21, 2021-12-28:
Well, it's named YI IoT Outdoor Camera - Model: XY-R9520-V3, it's wifi only. Camera is currently unpaired and pairing process fails with timeout. I.e. I use the App to generate the QR-code, camera scans it and successfully connects to Wifi but times out afterwards. I can ping the camera during that process. But since pairing fails, I assume that Wifi credentials are not stored.

cjj25, 2021-12-28:
That's correct, the `Factory` folder disables the Yi from performing certain actions such as pan/tilt when detected.

Does the backup folder exist on your SD card? This will give us a clue to how far the code is actually executing.

When the camera timeouts after pairing, if you restart the camera, does it connect back to WiFi (able to ping)?

mgerald21, 2021-12-28:
No, after the timeout of pairing and restart the camera does not reconnect to Wifi (thus unable to ping). It starts again with this beep-sequence for pairing.

No backup folder, I even doubt that config.sh is executed.

cjj25, 2021-12-28:
Interesting, I've just tried the same on a camera here and had the same issue. I wonder if this is a Yi service problem.

Try the other known hack for telnet here: https://github.com/cjj25/yi-hack-telnet-root

You'll need to edit the WiFi details

mgerald21, 2021-12-28:
Tried another 2GB and 8GB SD card with cjj25/yi-hack-telnet-root - No_Cloud_Connect_To_WiFi.
Strange thing it always starts with this beep-sequence for pairing and of course does not connect to Wifi

mgerald21, 2021-12-29:
Finally managed to enable telnet on the device *hooray* 
It seems that only higher class SD cards are supported and all my older ones were ignored.

mgerald21, 2021-12-29:
Alright, so I can login now but there is something wrong with the "stream", since there are core-dumps in the SD card root.
core.stream.580

```
/var/tmp/sd # export LD_LIBRARY_PATH=/lib:/home/lib:/home/rt/lib:/home/app/locallib:/var/tmp/sd/lib
/var/tmp/sd # ./Yi/load_cpld_ssp 
hw_ver[32]=[21131113100000000000000000000000] 
gpio_pin[32]=[11111111110011000000000000000000] 
moto not support !!!
/var/tmp/sd # ./stream
Segmentation fault (core dumped)
/var/tmp/sd #
```

mgerald21, 2021-12-29:
Ok, main issues solved. Closed

cjj25, 2021-12-29:
> Alright, so I can login now but there is something wrong with the "stream", since there are core-dumps in the SD card root. core.stream.580
>
> ```
> /var/tmp/sd # export [...]

Thanks for the feedback.

Can you confirm you managed to get the RTSPServer running in the end then?

mgerald21, 2021-12-30:
Yes, to summarize my findings.

Because camera is unpaired there are no Wifi credentials stored and wifi/config.sh is not executed.

Only the https://github.com/cjj25/yi-hack-telnet-root seems to work. For that I had to move the files into the Factory directory.
```
/Factory/config.sh
/Factory/factory_test.sh
/Factory/fork_process.sh
/Factory/make_backup.sh
/Factory/wpa_supplicant.conf
```

closed 2021-12-30 by mgerald21

cjj25, 2021-12-30:
Thanks for the update.

The earlier releases actually used the payload you're using now. It would be interesting to know if they work out the box for you.

Great work!