code-explorer1234/Yi-RTS3903N-RTSPServer #17: Telnet ok but no rtsp

vinceHH, 2022-01-06:
Hello,

Thanks for the hack, I had lost hope of his availability :)

My model is XY-R9520-V3

I'm able to connect to the camera using telnet

But when I'm trying to conect to the rstp video stream with VLC : 
Error messages :  
```
live555 error: Failed to connect with rtsp://192.168.1.151:554/ch0_0.h264
main error: connection failed: Connection refused by peer
satip error: Failed to connect to RTSP server 192.168.1.151:554
main error: connection failed: Connection refused by peer
access_realrtsp error: cannot connect to 192.168.1.151:554
```

any idea ?

Regards

cjj25, 2022-01-06:
Take a quick look at [this](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/10#issuecomment-942439342)

It sounds very similar to what you're experiencing. 

However, it looks like the port isn't open so the service might not be running.

Can you provide the output of `ps` when you login via telnet, a firmware dump will help tool (should be on the root of your SD in "backup")

vinceHH, 2022-01-06:
hello,

here the output of ps : 
`  PID USER       VSZ STAT COMMAND
    1 root      1428 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW<  [khelper]
   11 root         0 SW<  [writeback]
   12 root         0 SW<  [bioset]
   13 root         0 SW<  [kblockd]
   14 root         0 SW   [khubd]
   15 root         0 SW   [kworker/0:1]
   16 root         0 SW<  [cfg80211]
   17 root         0 SW   [kswapd0]
   18 root         0 SW   [fsnotify_mark]
   19 root         0 SW<  [crypto]
   28 root         0 DW   [enable_swp_task]
   47 root         0 SW<  [deferwq]
   48 root         0 SW   [kworker/u2:1]
   49 root         0 SW<  [kworker/0:1H]
  230 root      1428 S    syslogd
  232 root      1420 S    klogd
  237 root      1428 S    telnetd
  238 root         0 SWN  [jffs2_gcd_mtd5]
  241 root      1428 S    init
  405 root         0 SW   [mmcqd/0]
  436 root         0 SW   [RTW_CMD_THREAD]
  487 root       772 S    ./log_server
  507 root      9460 S    ./dispatch
  510 root      7520 S    ./mp4record
  511 root      8800 S    ./oss
  512 root     79720 S    ./rmm
  513 root      1084 S    ./watch_process
  516 root      1420 S    watchdog -t 2 -T 5 /dev/watchdog
  719 root      1284 S    wpa_supplicant -c/tmp/wpa_supplicant.conf -g/tmp/wpa
  813 root      1428 S    udhcpc -i wlan0 -b -s /backup/script/default.script
  817 root      1072 S    ./arp_test
  874 root     11540 S    ./cloud
  876 root     17996 S    ./p2p_tnp
 1463 root      1432 S    -sh
19407 root      1428 R    ps
`
for the backup, using telnet, I can see the folder : 
`~ # ls
backup   dev      home     lib      mnt      proc     root     tmp      var
bin      etc      init     media    overlay  rom      sys      usr
~ # cd backup/
/backup # ls
8188fu.ko         cloudAPI          init.sh           rsa_pub_dec       wifi_conf
ca                download_upgrade  mtd_img           script
/backup #
`

but, when I plug the SD card on my computer (Windows) , there is not backup folder on my SD card :(

Can I mount the SD card and copy the backup on it ?

cjj25, 2022-01-06:
This is very strange as the main process is still running, as if the payload on the SD card hasn't been picked up.

Do you have a log.txt on your sd card (if so, pls provide the output here)? 

What's the spec of your SD card, it has to be a [higher class as per this users report](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/16#issuecomment-1002462321)

vinceHH, 2022-01-06:
It's quite confusing for me.
I've made a simple test that running the camera without any SD card. 
And the result is the same, the camera is connected to the Wifi and I'm able to telnet with root and no password.
So, it ignore the SD card. I try with another one (HC 1), same result.

I have owned this camera for a long time. I remember pairing it with the app but that's not what I want to use it for. I only want to use rtsp stream

vinceHH, 2022-01-06:
I have another camera, brand new, out of the box, same model.
I try the 'simple' version of your hack to just enable telnet on it, with no success, camera is waiting for pairing.

I've make the test with 4 SD card, all are high class from 8 GB to 64 GB, same result.
I'm thinking that I'm doing something wrong ..

cjj25, 2022-01-07:
What's the model of both cameras? 

The camera that has telnet open, can you do `cd /var/tmp/sd` then `ls` does this list the contents of your sd card?

vinceHH, 2022-01-07:
both camera are  XY-R9520-V3 model

In /var/tmp/sd, I only have log.txt, log_oss.txt and the record folder

cjj25, 2022-01-07:
Could you provide a copy of the log.txt and log_oss.txt, can you confirm they're on the connected SD card?

vinceHH, 2022-01-07:
no, they are not stored on the connected SD card, it seems to be local storage

cjj25, 2022-01-07:
Well that confirms that it can't see the SD card. 

In telnet, what's the output of `dmesg` or `cat /var/log/messages`

Is the SD card formatted with FAT32?

vinceHH, 2022-01-07:
yes, the card is formatted with FAT32
I also make a try using the HP format tool.

here the output of dmesg : 
```
/var/tmp/sd # dmesg
Freeing unused kernel memory: 176K (803e4000 - 80410000)
usb 1-1: new high-speed USB device number 2 using ehci-platform
usb 1-1: New USB device found, idVendor=0bda, idProduct=f179
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: 802.11n
usb 1-1: Manufacturer: Realtek
usb 1-1: SerialNumber: 00e04c000001
jffs2: notice: (236) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
rlx snd internal codec init
soc-audio soc-audio.0.auto: ASoC: machine RLX_INTERN_CARD should use snd_soc_register_card()
soc-audio soc-audio.0.auto:  rlx-codec-digital <-> pcm-platform mapping ok
soc-audio soc-audio.0.auto:  rlx-codec-analog <-> pcm-platform mapping ok
usbcore: registered new interface driver rtl8188fu
rtscam:isp resvd mem addr : 0x00520000, size : 0x1900000
rtscam:rtscam_mem_init v:0xa0520000 p:0x00520000 s:0x00001900
rtscam:rtscam_lock_init
rtscam:rtscam_soc_probe
rtscam:rtscam_hx280_probe
rtscam:hx280enc:HW at base <0x18060000> with ID <0x48317011>
rtscam:rtscam_jpgenc_probe
rtscam:rtscam_osd2_probe
rtscam:rtstream_init
mmc0: new ultra high speed SDR50 SDHC card at address 1388
mmcblk0: mmc0:1388 USD00 14.8 GiB
 mmcblk0: p1
rtscam:begin to load fw from isp.fw
rtscam:Load firmware size : 131072.
rtscam:Found ISP 1.011 device
rtscam:video device <rts3903-isp> registered
rtscam:rts3903-isp initialized
rtscam:begin to load fw from isp.fw
rtscam:Load firmware size : 131072.
rtscam:begin to load fw from /home/lib/sc2230/isp.fw
rtscam:Load firmware size : 131024.
rtscam:Found ISP 1.011 device
rtscam:video device <rts3903-isp> registered
rtscam:rts3903-isp initialized
hw  =[21131113100000000000000000000000]
gpio=[11111111110011000000000000000000]
[CPLD_PERIPH] CPLD_PERIPH module inited
moto, stepmotor_function_init()

(strlen(hw)=(32), strlen(gpio)=(32)
hw[21131113100000000000000000000000]
gpio[11111111110011000000000000000000]
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 19
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 17
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 16
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 15
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 11
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 14
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 22
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 18
successful init
Started watchdog timer
Started watchdog timer

[PID_LIST] pid_list_init ok, [ ver=May 27 2019, 16:21:04 ]
/var/tmp/sd #
```

The SD card seems to be well detected : 
```
mmc0: new ultra high speed SDR50 SDHC card at address 1388
mmcblk0: mmc0:1388 USD00 14.8 GiB
```

vinceHH, 2022-01-07:
here the log files
[log.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7828574/log.txt)
[log_oss.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7828578/log_oss.txt)

vinceHH, 2022-01-07:
Some other information about the sd card : 
```
~ # fdisk /dev/mmcblk0 -l

Disk /dev/mmcblk0: 15.9 GB, 15917383680 bytes
255 heads, 63 sectors/track, 1935 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

        Device Boot      Start         End      Blocks  Id System
/dev/mmcblk0p1               1        1936    15544288+  c Win95 FAT32 (LBA)
~ #
```

mgerald21, 2022-01-07:
Can you please check the mounts:

```
Escape character is '^]'.
(none) login: root
 _           _         _            
| |         | |       | |           
| |     ___ | |__  ___| |_ ___ _ __ 
| |    / _ \| '_ \/ __| __/ _ \ '__|
| |___| (_) | |_) \__ \ ||  __/ |   
|______\___/|_.__/|___/\__\___|_|   
                                    
                                    
~ # mount
rootfs on / type rootfs (rw)
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
ramfs on /var type ramfs (rw,relatime)
ramfs on /dev type ramfs (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
tmpfs on /media type tmpfs (rw,relatime,size=64k,mode=755)
/dev/mtdblock5 on /backup type jffs2 (rw,relatime)
/dev/mtdblock4 on /home type squashfs (ro,relatime)
/dev/mmcblk0p1 on /var/tmp/sd type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /home/app/localko type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
~ # 

```
and if SD is not mounted, try to manually mount it.
```
~ # mount -t vfat /dev/mmcblk0p1 /var/tmp/sd
~ # 
```

If that succeeds if would be great to backup the flash to the SD, because your firmware behaves quite different.

vinceHH, 2022-01-07:
not mounted at boot : 
```
~ # mount
rootfs on / type rootfs (rw)
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
ramfs on /var type ramfs (rw,relatime)
ramfs on /dev type ramfs (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
tmpfs on /media type tmpfs (rw,relatime,size=64k,mode=755)
/dev/mtdblock5 on /backup type jffs2 (rw,relatime)
/dev/mtdblock4 on /home type squashfs (ro,relatime)
```

but successfully mounted manually : 
```

~ # mount -t vfat /dev/mmcblk0p1 /var/tmp/sd
~ # cd /var/tmp/sd
/var/tmp/sd # ls
System Volume Information  lib                        wifi
Yi                         rRTSPServer                wpa_supplicant.conf
invert_adc_rename_me       stream
/var/tmp/sd #
```

I don't undestand

mgerald21, 2022-01-07:
Can you please manually run the make_backup.sh script from the wifi directory to backup your firmware. Please zip the backup folder on your SD card and attach this archive to this issue.
```
~ # /var/tmp/sd/wifi/make_backup.sh
...
~ # 
```
We would like to look into that firmware, if it's newer or older to the known one and why the SD card is not mounted during the boot.

vinceHH, 2022-01-07:
here it is : 
[backup.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7828814/backup.zip)

thanks for your support

cjj25, 2022-01-07:
`not mounted at boot` so you're manually mounting the SD card?

In `/home/app/init.sh` it calls `/home/base/tools/checkdisk` which is responsible for mounting the SD card.. it is very fussy about the type of SD card it will allow though.

I'll take a look at your firmware shortly.

@mgerald21 - thanks for helping out

cjj25, 2022-01-07:
`7.1.00.11A_201905311440` appears to be a supported firmware now pointing more towards the SD card.

Try the following with the SD card mounted.

```
cd /var/tmp/sd/wifi
./config.sh

```

What's the output?

cjj25, 2022-01-07:
I'd also try the following on a freshly booted camera.

`/home/base/tools/checkdisk` - what's the output?

vinceHH, 2022-01-07:
here the output of config.sh : 

```
/var/tmp/sd/wifi # ./config.sh
Telnet already running
udhcpc (v1.20.2) started

Setting IP address 0.0.0.0 on wlan0
cp: can't create '/home/app/localko/cpld_periph.ko': Read-only file system
cp: can't create '/home/app/localko/pid_list.ko': Read-only file system
cp: can't create '/home/app/localko/ssp_ms41909.ko': Read-only file system
cp: can't create '/home/app/localko/ssp_ms41909_union.ko': Read-only file system
Sending discover...
Sending select for 192.168.1.151...
hw_ver[32]=[21131113100000000000000000000000]
gpio_pin[32]=[11111111110011000000000000000000]
insmod: can't insert '/home/app/localko/cpld_periph.ko': File exists
moto not support !!!
Received SIGTERM
set H264 to CVBR mode.
killall: init.sh: no process killed
/var/tmp/sd/wifi #
```

and here the output of checkdisk : 

```
/home/base/tools # ./checkdisk
get_cluster_size(349) sector_size:512,sect_per_cluster:16
checkdisk, main() ret_status(10002)
```

vinceHH, 2022-01-11:
Hi,

Do you need other information ?

regards,

Vince

vinceHH, 2022-01-19:
Hi,

It's working ! 
Seems to be due to the format of the SD card.
Formatting with FAT32 with Windows don't work
Formatting with FAT32 with Linux using your command don't work too.
Formatting via the app is the working solution. I'm now able to get the rtsp stream.

Thanks again for your support !

cjj25, 2022-01-19:
Fantastic news! Thanks for confirming the solution. It helps solve any problems other may face in future :)

closed 2022-01-19 by cjj25