code-explorer1234/Yi-RTS3903N-RTSPServer #18: Just a question

fkoteam, 2022-01-11:
hi, 
Yi-RTS3903N-RTSPServer works for me very well. I'm thinking to install zoneminder or a similar software in a cloud (like oracle) but my connection is behind a cgnat and I don't have a public ip. Is it possible to achieve this? 

Thank you!

cjj25, 2022-01-11:
Great to hear it's working well!

I use BlueIris but it's hosted locally on a server on my LAN.

If you're thinking of hosting the DVR in the cloud (such as Zoneminder) then I wouldn't expose the camera directly to the internet. I think installing a VPN server such as OpenVPN on the cloud server, then setup OpenVPN on the camera (You'll be able to compile with the toolchain) so it connects to the cloud.

Depending on how advanced your router is, it might be able to run the VPN for you. Again, if you've got an advanced firewall / router such as PfSense then you could directly expose your camera on the network, use a service such as Dynamic DNS and restrict the access to the RTSP port to a single IP address (your cloud server).

I hope this helps!

fkoteam, 2022-01-11:
thank you for your time. Openvpn client has a lot of dependencies and I think I wouldn't to be able to compile it with my knowledge :(  And as you say I'm afraid that the cpu's cam won't be able to run all services appropriately.
Only one more question (I promise!): in your research, have you discovered how the original firmware works? Does the cam send the stream to a yi server? is there any chance to emulate this strategy?

cjj25, 2022-03-16:
I did a fair amount of research in disassembling and reverse engineering the binaries for the Yi Camera. 

There's basically two ways it operates (for camera viewing)

1. LAN is detected, the app requests a token from the Yi servers, it then opens a port and listens for this token.. streaming is then done locally
2. P2P this method does the above but uses the Yi servers as a man-in-the-middle/proxy relay

Re: emulation
I've not checked to see if there's any certificate pinning but the difficult part is how the client (iOS/Android) uses the tokens / decrypts the stream

closed 2022-03-16 by fkoteam