code-explorer1234/Yi-RTS3903N-RTSPServer #19: Neither RTSPServer nor telnetd starting on a model XY-R9420-X2

iMilnb, 2022-01-13:
Hi there,

First, thanks a lot for working on this project!
So I have a Cacagoo webcam labelled `model XY-R9420-X2` running firmware `7.1.00.25A_202002271051`, I formatted a 2GB sdcard with FAT32, untarred the latest `tar.gz` (Yi-RTS3903N-RTSPServerV03.tar.gz) and fired up the cam with this content, but nothing happened really. The card doesn't hold any dump either, nothing happens, the cam boots, does its init and works in a standard way. While at it I tried with V02 and V01 also, same result.  
I probably am missing something obvious here, but after a couple hours trying, I can't seem to know what.

iMilnb, 2022-01-13:
Well, it finally worked.  
What I did was formatting the card with the camera itself, then mounted the partition using plain `sudo mount`, I think user automount does permission trickery that the camera doesn't like.

closed 2022-01-13 by iMilnb

cjj25, 2022-01-13:
That's great news! Could you post your exact solution?

This user appears to be having the problem: https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/17

iMilnb, 2022-01-13:
> That's great news! Could you post your exact solution?
> 
> This user appears to be having the problem: #17

Sure!