code-explorer1234/Yi-RTS3903N-RTSPServer #21: Goowls 1080P IP Camera supported!

iMilnb, 2022-01-14:
Just pointing out that this camera https://www.newegg.com/p/2P0-02K4-00008 has exactly the same firmware as the Cacagoo XY-R9420-X2 (7.1.00.25A_202002271051) and thus the archive works out of the box.
Firmware backup is here: https://imil.net/stuff/goowls_dump.tgz

cjj25, 2022-01-14:
Thanks for the update and firmware :-)
Similar devices here: https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/5#issuecomment-903747367

closed 2022-01-14 by cjj25

Timmy-Harris, 2022-07-05:
@iMilnb Hello! Could you please share firmware backup? 'cause this link https://imil.net/stuff/goowls_dump.tgz does not work :(
Is there any way to update a very similar camera based on MVT9632-SC2303_V1.0 with your backup? 
Thank you in advance.

Timmy-Harris, 2022-07-05:
The vendor is Sdeter A8. I'm using V380 application.
[image]

cjj25, 2022-07-05:
When you use the payload from this repository on your SD card, does it do a firmware dump  (create a backup folder on the root of the SD card)?

Timmy-Harris, 2022-07-05:
Unfortunately, it does nothing. The device just started video recording and saved it into RecFiles folder.

cjj25, 2022-07-05:
I'd love to have a firmware dump to see if there is a payload we could create for other users. 

Do you have access to UART? 

[UART information here.](https://github.com/TheCrypt0/yi-hack-v4/issues/154#issuecomment-1038244399)

cjj25, 2022-07-05:
https://github.com/TheCrypt0/yi-hack-v4/issues/154#issuecomment-903343524

Timmy-Harris, 2022-07-06:
Yes, I have UART converter. I'll try to connect this board.

Timmy-Harris, 2022-07-07:
I've got the same behavior as [bondiblueos9](https://github.com/bondiblueos9) got. I wired up those pads but got nothing. I tried the tx and rx both ways. Checked two middle pins and those pins connected to the RTS3903N chip.

Ammarpic, 2022-07-12:
> Just pointing out that this camera https://www.newegg.com/p/2P0-02K4-00008 has exactly the same firmware as the Cacagoo XY-R9420-X2 (7.1.00.25A_202002271051) and thus the archive works out of the box.
I have a problem with the camera
I need the basic software
Please help