code-explorer1234/Yi-RTS3903N-RTSPServer #22: Wifi doesnt work

bart6661, 2022-01-18:
Hey 
i have two YI Outdoor Cameras with RTS3903N. 
The first work perfect, but without wpa_supplicant.conf (it saves the wifi settings from qr code...)
the second had a Problem with loading wpa_supplicant.conf 
The camera doesn't connect to wifi at start, after starting i scan the QR Code from the APP and then its connect to wifi. 
Telnet and stream is working but no automatic wifi connect.
Thats my actual wpa_supplicant.conf File. 
[wpa_supplicant.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/7892117/wpa_supplicant.txt)

what is my mistake? 
I hope you can help me :/
thanks Bart

cjj25, 2022-01-19:
Hi Bart,

There appears to be a few reports about the pairing process failing (#15) and my stock cameras are also failing to pair.

I've noticed there is a typo in the documentation (my bad), can you make sure the wpa_supplicant.txt file is stored in the[ path being checked](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/7763840399c08659c6b2e875bc4bca5a30f7f58b/build/sd_payload/wifi/fork_process.sh#L19)  `ROOT_OF_SD_CARD/Factory/wpa_supplicant.conf`

If above doesn't help, continue below.

Unfortunately the current payload we're using requires for the initial WiFi to have been setup BEFORE its able to run our payload.

However, I'm sure the earlier version did work with cameras that hadn't been paired previously.

It's worth heading over to the [releases page](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/releases) and trying v01 and v02 as they used a slightly different payload. 

If the camera boots up and you don't hear the beeping sound, that's good news.. the payload has somewhat worked and is pointing to the wpa_supplicant.txt file.

If all that fails, try using the wpa_supplicant.txt file and the simple telnet hack here: [cjj25/yi-hack-telnet-root](https://github.com/cjj25/yi-hack-telnet-root)

bart6661, 2022-01-20:
WTF... i try so many thinks and nothing works, and today i try again, and wow  the  yi-hack-telnet-root works ... at the first try... After this i test your Version 2 and copy the wifi part to your start_process.. and delete your part... 
```
wpa_supplicant -c/var/tmp/sd/Factory/wpa_supplicant.conf -g/tmp/wpa_supplicant-global -Dwext -iwlan0 -B
#
## Wait for the connection
sleep 10s
#
## Setup the network and wait for it
udhcpc -i wlan0 -b -s /backup/script/default.script
sleep 10s
```
Now it works fine :) thanks for your help...

closed 2022-01-20 by bart6661