code-explorer1234/Yi-RTS3903N-RTSPServer #24: Help for Tuya Camera

ifunboy, 2022-05-17:
Hi!
I've a Tuya Camera(mode:SC-WA002) .The soc is RTS3903N.I try to extract the V3 files to the root of a MicroSD,but not works and it seems boot from the flash(Not found the ttl port so I guess).I dump the flash by spiflash debuger.
A lot of Thanks！
[tuya-cam.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/8709961/tuya-cam.zip)
[image]
[image]
[image]
[image]
[image]
[image]
Soc:RTS3903N
Flash:cFeon QH128A-104HIP
Wifi:RTL8188

cjj25, 2022-05-17:
I've been waiting to get my hands on the tuya firmware that runs on this chipset. 

Thanks for providing the dump, I'll take a look and see if I can find a way in!

cjj25, 2022-05-17:
On first glance it looks like the following 

`ty_sdcard_check_upgrade.sh` is looking for directory `/tuya/upg` to exist on the sdcard, it wants a hidden text file in this directory called `.tyupg.cfg` with the value `1` inside file.

This then calls a script named `ty_sdcard_upgrade.sh` which should be created at `/tuya/upg/ty_sdcard_upgrade.sh`

You should be able to create a back door to start telnet with this. However, as you've got SPI access it might be easier to just edit the `/etc/init.d/rcS` and uncomment the telnetd line.

```
#! /bin/sh

MNTDIR=$1
UPDDIR=${MNTDIR}/tuya/upg
IFUPD=0

check_if_need_upgrade()
{
	if [ ! -d ${UPDDIR} ]; then
		echo "*****[upgrade info]***** no upgrade dir, no need to check..."  > /dev/console
		exit
	fi

	upgcfg=${UPDDIR}/.tyupg.cfg

	if [ -e ${upgcfg} ]; then
		forcestu=`cat ${upgcfg}`

		if [ x"${forcestu}" = x"1" ]; then
			IFUPD=1
			echo "*****[upgrade info]***** force to upgrade..."  > /dev/console
			return
		fi
	fi
}

if [ x"${MNTDIR}" = x"" ]; then
	echo "*****[upgrade info]***** no dir param, no need to check..."  > /dev/console
	exit
fi

sleep 0.6

check_if_need_upgrade

if [ x"${IFUPD}" = x"1" ]; then
	if [  -e ${UPDDIR}/ty_sdcard_upgrade.sh ]; then
		echo "*****[upgrade info]***** start to upgrade..."  > /dev/console
		[ -x ${UPDDIR}/ty_sdcard_upgrade.sh ] && ${UPDDIR}/ty_sdcard_upgrade.sh
	else
		echo "*****[upgrade info]***** no upgrade script, check it..."  > /dev/console
	fi
else
	echo "*****[upgrade info]***** no need upgrade, check cfg no set..."  > /dev/console
fi

```

ifunboy, 2022-05-17:
I buy it from taobao. I will go on tomorrow. Thanks.
The Amazon Link like this [https://www.amazon.com/LARKKEY-Wireless-Security-Surveillance-Compatible/dp/B07X94S79C](url)

cjj25, 2022-05-17:
You're welcome.

I've also scanned the main binary and I can see there looks to be a built in RTSP server for "echoshow_task" - you could possibly patch the binary or use my libc from the tools repo to hook into this function.

Let me know how you get on!

ifunboy, 2022-05-19:
I had replaced the /etc/shaow and enabled the telnet.
Now I'm trying to start the rtsp server.
I copy the file to tf card.And export the LD_LIBRARY.At last,I can get the tool to work with raw (?) and jpeg images, but not with h264 compressed.Just like [https://github.com/cjj25/RTS3903N-Tools/issues/1](url).
V3 file
`/mnt/mmcblk0p1/rts3903 # ./dbg_isp --preview --max 5 --number 5 --save /mnt/sdcard/ --show --h264
dbg_isp log: Current fmt is : NV12 1920x1080
dbg_isp log: Current fps is : 1/20
dbg_isp error: H264EncInit failed
dbg_isp error: init h264 env fail
dbg_isp log: init_buffers
dbg_isp log: reqbufs, count = 2
dbg_isp log: buffer count = 2
dbg_isp log: stream on 
dbg_isp log: Get frame at [1652978231, 253163]
dbg_isp log: Get frame at [1652978231, 520952]
dbg_isp log: Get frame at [1652978231, 865444]
dbg_isp log: Get frame at [1652978232, 157079]
dbg_isp log: Get frame at [1652978232, 458106]
dbg_isp log: Get frame at [1652978232, 794120]
dbg_isp log: release_bufs
dbg_isp log: Cmd (preview) success. Use time:1989`

extracted.tar.gz from RTS3903N-Tools issue #1
`/mnt/mmcblk0p1/rom/bin # ./dbg_isp --preview --max 5 --number 5 --save /mnt/sdcard/ --show --h264
dbg_isp log: Current fmt is : NV12 1920x1080
dbg_isp log: Current fps is : 1/20
dbg_isp log: ++++++++++++++++++++++++++++++++
dbg_isp log: H264EncInit#
dbg_isp log: streamType : 0
dbg_isp log: viewMode : 1
dbg_isp log: level : 40
dbg_isp log: refFrameAmount : 1
dbg_isp log: width : 1920
dbg_isp log: height : 1080
dbg_isp log: frameRateNum : 20
dbg_isp log: frameRateDenom : 1
dbg_isp log: scaledWidth : 0
dbg_isp log: scaledHeight : 0
dbg_isp log: H264EncSetCodingCtrl#
dbg_isp log: sliceSize : 0
dbg_isp log: seiMessages : 0
dbg_isp log: videoFullRange : 0
dbg_isp log: constrainedIntraPrediction : 0
dbg_isp log: disableDeblockingFilter : 0
dbg_isp log: sampleAspectRatioWidth : 0
dbg_isp log: sampleAspectRatioHeight : 0
dbg_isp log: enableCabac : 1
dbg_isp log: cabacInitIdc : 0
dbg_isp log: transform8x8Mode : 2
dbg_isp log: quarterPixelMv : 2
dbg_isp log: cirStart : 0
dbg_isp log: cirInterval : 0
dbg_isp log: intraSliceMap1 : 0
dbg_isp log: intraSliceMap2 : 0
dbg_isp log: intraSliceMap3 : 0
dbg_isp log: intraArea.enable : 0
dbg_isp log: intraArea.top : 68
dbg_isp log: intraArea.bottom : 68
dbg_isp log: intraArea.left : 120
dbg_isp log: intraArea.right : 120
dbg_isp log: roi1Area.enable : 0
dbg_isp log: roi1Area.top : 68
dbg_isp log: roi1Area.bottom : 68
dbg_isp log: roi1Area.left : 120
dbg_isp log: roi1Area.right : 120
dbg_isp log: roi2Area.enable : 0
dbg_isp log: roi2Area.top : 68
dbg_isp log: roi2Area.bottom : 68
dbg_isp log: roi2Area.left : 120
dbg_isp log: roi2Area.right : 120
dbg_isp log: roi1DeltaQp : 0
dbg_isp log: roi2DeltaQp : 0
dbg_isp log: adaptiveRoi : 0
dbg_isp log: adaptiveRoiColor : 0
dbg_isp log: fieldOrder : 0
dbg_isp log: gdrDuration : 0
dbg_isp log: H264EncSetRateCtrl#
dbg_isp log: pictureRc : 1
dbg_isp log: mbRc : 0
dbg_isp log: pictureSkip : 0
dbg_isp log: qpHdr : -1
dbg_isp log: qpMin : 10
dbg_isp log: qpMax : 42
dbg_isp log: bitPerSecond : 2097152
dbg_isp log: gopLen : 30
dbg_isp log: hrd : 0
dbg_isp log: hrdCpbSize : 0
dbg_isp log: intraQpDelta : 0
dbg_isp log: fixedIntraQp : 0
dbg_isp log: mbQpAdjustment : 0
dbg_isp log: longTermPicRate : 15
dbg_isp log: mbQpAutoBoost : 0
dbg_isp log: H264EncSetPreProcessing#
dbg_isp log: origWidth : 1920
dbg_isp log: origHeight : 1080
dbg_isp log: xOffset : 0
dbg_isp log: yOffset : 0
dbg_isp log: inputType : 1
dbg_isp log: rotation : 0
dbg_isp log: videoStabilization : 0
dbg_isp log: colorConversion.type : 1
dbg_isp log: colorConversion.coeffA : 13933
dbg_isp log: colorConversion.coeffB : 46871
dbg_isp log: colorConversion.coeffC : 4732
dbg_isp log: colorConversion.coeffE : 35317
dbg_isp log: colorConversion.coeffF : 41615
dbg_isp log: scaledOutput : 0
dbg_isp log: interlacedFrame : 0
dbg_isp log: --------------------------------
dbg_isp log: init_buffers
dbg_isp log: reqbufs, count = 2
dbg_isp error: request bufs failed
dbg_isp error: init buffers failed
dbg_isp error: Stream on failed
dbg_isp error: Cmd (preview) fail. Use time:22
`

ifunboy, 2022-05-19:
Maybe make a mjpeg-streamer is great

cjj25, 2022-05-19:
Have you extracted the mtdblock4 and repacked it or did you use my sdcard trick? 

If you're confident in repacking the squashfs, I believe I can patch your file to run the RTSP directly from tuya?

ifunboy, 2022-05-20:
I tried to unpack the Yi-RTS3903N-RTSPServerV03.tar.gz and copied it to the tf card.Then i export the LD_LIBRARY from tf card.And run the dbg_isp. I can get the tool to work with raw and jpeg images, but not with h264 compressed

> /mnt/mmcblk0p1/rts3903 # ./dbg_isp --preview --max 5 --number 5 --save /mnt/sdcard/ --show --h264
dbg_isp log: Current fmt is : NV12 1920x1080
dbg_isp log: Current fps is : 1/20
dbg_isp error: H264EncInit failed
dbg_isp error: init h264 env fail
dbg_isp log: init_buffers
dbg_isp log: reqbufs, count = 2
dbg_isp log: buffer count = 2
dbg_isp log: stream on
dbg_isp log: Get frame at [1652978231, 253163]
dbg_isp log: Get frame at [1652978231, 520952]
dbg_isp log: Get frame at [1652978231, 865444]
dbg_isp log: Get frame at [1652978232, 157079]
dbg_isp log: Get frame at [1652978232, 458106]
dbg_isp log: Get frame at [1652978232, 794120]
dbg_isp log: release_bufs
dbg_isp log: Cmd (preview) success. Use time:1989

ifunboy, 2022-05-20:
I modified the ty_monitor.sh and comment the reboot line so I can kill the ty_cam to stop tuya services.

cjj25, 2022-05-20:
Replace your tycam (make a backup first) with this patched version.

/dev/mtdblock4 - /app/skyeye/bin/tycam

It should hopefully create an RTSP server on port 8554 if it works.

Make sure no other binary that accesses the camera starts first, it needs exclusive access to its ring buffer and the video driver.
[tycam_patched.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/8738512/tycam_patched.zip)

I've not tested it so do let me know the outcome, it might be limited to the SD stream and not the HD.

Could you also provide a picture of the UART ports you found to help future users?

ifunboy, 2022-05-20:
The UART port
[image]

ifunboy, 2022-05-20:
I Replaced tycam in /dev/mtdblock4 ,but don't work.
I see the dh -f result
`/dev/mtdblock6 2.8M 2.8M 0 100% /opt`
Maybe the system runs a script to the mount point from mtdblock4 to mtdblock6.（Maybe ty_upgrade.sh）
So I Replaced tycam in /dev/mtdblock6 and the rtspserver works！This is awesome!
But the resolution is locked at 640x360.Anyway, it's a big development.
The tycam from mtdblock6 is biger than mtdblock4.
[tycam.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/8742209/tycam.zip)
Next, it would be great to increase the video resolution and block the tuya network.

cjj25, 2022-05-20:
I have a solution for you, I'll be posting shortly :)

cjj25, 2022-05-20:
Please take a look at [cjj25/RTS3903N-Tuya-RTSPServer](https://github.com/cjj25/RTS3903N-Tuya-RTSPServer) and let me know how you get on

cjj25, 2022-05-24:
@ifunboy any update?

ifunboy, 2022-05-24:
Excuse my late reply,I'm busy with other things.
Talk about progress.
1.The RTS3903N-Tuya-RTSPServer can not work.
It seems the ty_sdcard_upgrade.sh not work.
The telnetd does not start，and can't connect to wifi.
I try to run the start_patched_tuya.sh manually，it did't work too，because when the ty_wdt was killed the machine restarted.And the ty_monitor.sh killed the telnetd.
`count=0
wifi_set_flag=0
echo "kill telnetd"
killall telnetd
while [ true ]`
So I replace the tycam_patched by SPI-flasher,very happy it works well both  Low Res or High Res.But there is still a problem.when I block the camera to accessing the Internet in router.The camera start to reconnect the wifi and stop the rtsp.So I try to compile the RTSPServer.The good news is it work with dgb_isp and rTSPServer.I edited the ty_startup.sh
`#!/bin/sh
export LD_LIBRARY_PATH=/lib:/opt/lib:/usr/local/lib:/usr/lib
export PATH=/usr/bin:/usr/sbin:/bin:/sbin:/opt/bin/:/opt/skyeye/bin/
sleep 5

wpa_supplicant -B -i wlan0 -c/mnt/sdcard/wpa_supplicant.conf -Dwext -dd
echo 0x00  > /proc/net/rtl8188fu/wlan0/chan_plan
udhcpc -b -i wlan0
sleep 3
telnetd&
echo 1 > /sys/devices/platform/rts_soc_camera/loadfw
/mnt/sdcard/startup.sh`
The root password is 12345
And start the rTSPServer by startup.sh
I bought a sop8 socket.it's much more convenient.
[image]
The next work is to find out why ty_sdcard_upgrade.sh didn't work.
And is there anyway to block the tycam to reconnect the wifi.
Thank You！:)

closed 2022-06-29 by cjj25

asanfang, 2022-07-08:
@ifunboy 你好，你的SC-WA002刷Tuya RTSP成功了吗？可不可以跳过涂鸦后台运行?

cjj25, 2022-07-08:
@asanfang Please use English when replying to tickets, you'll receive support from more people this way.