code-explorer1234/Yi-RTS3903N-RTSPServer #27: Works with Apeman IH74

ptersilie, 2022-06-23:
I can confirm that this works with the Apeman IH74 (which I believe is also sold as Voger VP230 and possibly others).
[image]

Firmware: 7.1.00.19A_201910181012

The only issue I encountered was that the RTSP stream by default was flipped upside down. But that was easily fixed by setting `RTS_VIDEO_CTRL_ID_FLIP` accordingly:

```diff
diff --git a/build/src/stream.c b/build/src/stream.c
index 3991826..1375e2c 100644
--- a/build/src/stream.c
+++ b/build/src/stream.c
@@ -249,6 +249,7 @@ int start_stream() {
     rts_av_enable_chn(h264);
     manage_modes(RTS_VIDEO_CTRL_ID_NOISE_REDUCTION, 100);
     manage_modes(RTS_VIDEO_CTRL_ID_LDC, 1);
+    manage_modes(RTS_VIDEO_CTRL_ID_FLIP, 1);
 //    manage_modes(RTS_VIDEO_CTRL_ID_FOCUS, 100);
     manage_modes(RTS_VIDEO_CTRL_ID_DETAIL_ENHANCEMENT, 100);
     manage_modes(RTS_VIDEO_CTRL_ID_3DNR, -1);
```

cjj25, 2022-06-24:
Thanks for the feedback @ptersilie - Can you confirm your camera is Yi based and not Tuya? 

I actually have the Voger VP230 (physically identical) sitting in a box of cameras to help with my research, it's the RTS3903N but with Tuya's cloud software.  They sold for really cheap on Amazon for around £11 each.

There's a separate repo just for the Tuya implementation here: [cjj25/RTS3903N-Tuya-RTSPServer](https://github.com/cjj25/RTS3903N-Tuya-RTSPServer)

ptersilie, 2022-06-24:
The firmware of the camera is `7.1.00.19A_201910181012` and the scripts in this repo worked, which leads me to believe it is indeed Yi based. Is there a way to tell for sure?

Also while I have you here. Do you know what arguments I need to use for `RTS_VIDEO_CTRL_ID_ROTATE`. I noticed that `FLIP` wasn't enough as the image is then mirrored, so I would have to use `FLIP` and `MIRROR`. I reckon just using `ROTATE` might be easier, but I'm not sure if it expects an angle e.g. `180` or a direction e.g. `3` for 180 degrees.

ptersilie, 2022-06-24:
Thanks! I'll have a look later. Do you know where the binaries in https://github.com/cjj25/RTS3903N-Tools/tree/master/Realtek%20Specific come from? Do you have a source for those or are they from some kind of closed SDK?

cjj25, 2022-06-29:
I'm closing this issue for now - please feel free to add any discoveries you make regarding the parameters :)

ptersilie, 2022-06-30:
Hey, sorry, I didn't get around looking into this further yet, but will update this issue when I do. And thanks again for your help! :-)