code-explorer1234/Yi-RTS3903N-RTSPServer #29: Firmware rtsp3903n+45 camera from NL

McPrapor, 2022-10-25:
I found you got a similar board as I (https://user-images.githubusercontent.com/31572463/136786404-db857b44-ab6a-4585-aebe-1f5aa3b5d77f.jpg) could you please let me know which pins you used for uart? I'd gladly share the firmware, if it will be possible to access uart.
[image]

McPrapor, 2022-10-25:
Found this one as well https://github.com/TheCrypt0/yi-hack-v4/issues/154 I tried those those pads in the middle both ways and no output on my CP2102 showed.

cjj25, 2022-10-25:
Can you post the exact model of this camera and maybe an online shop url? 

It'll give me a clue as to what firmware its running and if I know of any known payloads you could try via the SD.

McPrapor, 2022-10-26:
It's Kruidvat Smart

https://www.kruidvat.nl/kruidvat-smart-indoor-ip-camera/p/5314128?gclid=Cj0KCQjwteOaBhDuARIsADBqRehNUKIbFA0hlQAVy9L2kdPxGX0-fsCEMcw0oAUh52urPL-edkCT__AaAssuEALw_wcB

I tried some sdcard payloads like Mercury720/1080 but interesting part
everything I put on sdcard being removed during camera boot process. I'll
investigate more and report.

McPrapor, 2022-10-26:
There is "Kruidvat Smart Home" Android app for control. Device info in the
app says that "Main module" and "MCU module" version is 5.2.7.

There is a manual book in the box which says the exact model is "SH.63.224".
Here is the link on pdf, though it's in Dutch and there is nothing useful
IMHO.
https://handleidingkwijt.com/kruidvat-sh-63-224-ip-camera/

cjj25, 2022-10-26:
This sounds like a Tuya based camera.

Try my payload on the [Tuya repo](https://github.com/cjj25/RTS3903N-Tuya-RTSPServer) and report back :)

cjj25, 2022-11-25:
Did you try this method?

McPrapor, 2022-11-25:
Yes, I tried with all binaries combinations, unfortunately no luck, nothing
changes. I tried as well mercury1080p solution without any luck. Going to
try LSC solution from Guino next week, just found new discussion, I have
those cams as well, maybe it will work for Kruidvat.

What HW could I use to dump firmware in case of failure? I tried to find an
flash memory chip, but quick look didn't give any result. Any suggestion?

cjj25, 2022-11-25:
For taking a dump of your SOIC8 chip, [something like this](https://amzn.eu/d/duR98jg) should work for you.

I use the same board as this camera for my development, but it was originally a Yi based camera.

When you tried my Tuya RTSP method, did it create a backup folder on the root of your SD card?

McPrapor, 2022-11-25:
No backup folder. It doesn't run the scripts during the startup, looks like
something differs in the firmware. And it shows pretty strange fw version
in the app. Thanks for advice, I'll do more tests and try to take a dump,
if there will be no success.

cjj25, 2022-11-25:
It's certainly a firmware I've not seen before!  I'd love to have a copy to create a payload for it.

Keep us in the loop!

McPrapor, 2022-11-26:
So no luck with other payload methods I found, going to proceed with the
programmer. I believe this one is same as you posted?
https://www.amazon.nl/dp/B08TVNPTQK/

cjj25, 2022-11-26:
That should do the trick.

Do you have access to a soldering iron? I remove my chips before reading them as attempting while still soldered to the PCB often fails.

McPrapor, 2022-12-15:
https://github.com/guino/LSC1080P here it describes how to get working telnet without breaking the device, it's enough to put product.cof file with specific text on sdcard.

cjj25, 2022-12-15:
> So bingo, I got it. 8MB of pure happiness. binwalk says there is tons of interesting inside. Here you go @cjj25 : https://mega.nz/folder/eeZk0QSB#HUh2Q7p-OytEX52cmvk68g
> 
> I'll leave chip detached(kh25l6436f) for a while, so let me know, if there something else I could dump from it.

Welcome to the world of tinkering... this might be the start of dumping many flash chips! 

Your firmware appears to be a hybrid of this [project (dgiot)](https://github.com/dgiot/dgiot).

Interestingly link you posted, I've checked your firmware and confirm there are references in the main binary for that file.. specifically:

- `/mnt/mmc/mmc1/product.cof`
- `/etc/conf/product.cof`

These are the default settings cooked into your firmware:  `/etc/conf/product.cof

````
[CONST_PARAM]
lamp_board=0
capture_vol=100
play_vol=100
ptz_opposite_run=2
image_flip=0
ircut_flip=0

[DEFAULT_SETTING]
telnet=0
language=1100

[SENSITIVE]
day_num=0
night_num=0

````

The Tuya IPC SDK that's cooked into the binary does contain the RTSP ability and can most likely be patched the same way as I do the other binaries, specifically between 0x0047b544 (tuya entry) and 0x007bd1c4 (start streams).

Have you tested enabling telnet with the method you posted? 

There's also references to it in the `/etc/init.d/ (commented out)

McPrapor, 2022-12-15:
I was able to get telnet access, tried to find any possible secret option to enable rtsp in an easy way, but no luck so far.

cjj25, 2022-12-15:
Please see this issue posted on my[ other repo (exact same firmware)](https://github.com/cjj25/RTS3903N-Tools/issues/1) and how he managed to use my other RTSPServer.

Also, interesting WiFi defaults:
SSID: TP-LINK_DG
Password: dgiot0202

cjj25, 2022-12-15:
> I was able to get telnet access, tried to find any possible secret option to enable rtsp in an easy way, but no luck so far.

I've not flashed your firmware to my dev camera yet.

Could you give me the output of the following command "cat /proc/mtd"  and "dmesg"

Then possibly run the [make_backup.sh script](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/master/build/sd_payload/wifi/make_backup.sh) on an SD (you'll need to modify in the script paths from /tmp/sd/ to /mnt/mmc/mmc1/) so we have all the individual blocks and their correct sizes (makes it easier rebuilding the firmware parts).

McPrapor, 2022-12-15:
`~ # cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00800000 00010000 "global"
mtd1: 00040000 00010000 "boot"
mtd2: 00010000 00010000 "hconf"
mtd3: 001c0000 00010000 "kernel"
mtd4: 00540000 00010000 "rootfs"
mtd5: 000b0000 00010000 "userdata"`
`~ # dmesg
turn off boot console early0
hconf type 3, size 0x10000, ebsize 0x10000
is_header invalid header: magic 0xffffffff, num -1, len -1
hconf_init hconf init success
ALSA device list:
  No soundcards found.
Warning: unable to open an initial console.
VFS: Mounted root (squashfs filesystem) readonly on device 31:4.
Freeing unused kernel memory: 180K (80483000 - 804b0000)
usb 1-1: New USB device found, idVendor=0bda, idProduct=f179
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: 802.11n
usb 1-1: Manufacturer: Realtek
usb 1-1: SerialNumber: 68B9D34E5708
rlx snd internal codec init
soc-audio soc-audio.0.auto: ASoC: machine RLX_INTERN_CARD should use snd_soc_register_card()
soc-audio soc-audio.0.auto:  rlx-codec-digital <-> pcm-platform mapping ok
soc-audio soc-audio.0.auto:  rlx-codec-analog <-> pcm-platform mapping ok
rtscam:isp resvd mem addr : 0x005c0000, size : 0x157f000
mmc0: new SDHC card at address 0001
mmcblk0: mmc0:0001 SD8GB 7.28 GiB
 mmcblk0: p1
rtscam:rtscam_mem_init v:0xa05c0000 p:0x005c0000 s:0x0000157f
rtscam:rtscam_lock_init
rtscam:rtscam_soc_probe
rtscam:rtscam_hx280_probe
rtscam:hx280enc:HW at base <0x18060000> with ID <0x48317011>
rtscam:rtscam_jpgenc_probe
rtscam:rtscam_osd2_probe
rtscam:rtstream_init
usbcore: registered new interface driver rtl8188fu
jffs2: notice: (389) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
rtscam:begin to load fw from isp.fw
rtscam:Load firmware size : 131024.
rtscam:begin to load fw from /lib/firmware/SC1245.bin
rtscam:Load firmware size : 131024.
rtscam:Found ISP 1.011 device
rtscam:video device <rts3903-isp> registered
rtscam:rts3903-isp initialized
Started watchdog timer
Started watchdog timer
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 19
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 17
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 16
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 15
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 11
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 14
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 22
MS_MHal_GPIO_Pad_Oen: setting gpio_direction_output 18
gpio_init ---- ret = 0
ker_driver_init
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
rtscam:get regulator io fail
rtscam:get io power fail
rtscam:get regulator analog fail
rtscam:get analog power fail
rtscam:get regulator core fail
rtscam:get core power fail
(motor = 0,     time = 2500000
(motor = 1,     time = 3500000`

cjj25, 2022-12-15:
Awesome thanks, what's the output of the following:

- ps -a (or ps -aux)
- mount

McPrapor, 2022-12-15:
```
~ # ps au
ps: invalid option -- a
~ # ps -au
ps: invalid option -- a
~ # ps u
ps: invalid option -- u
~ # ps -u
ps: invalid option -- u
~ # ps
  PID USER       VSZ STAT COMMAND
    1 root      1244 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW<  [khelper]
   11 root         0 SW<  [writeback]
   12 root         0 SW<  [bioset]
   13 root         0 SW<  [kblockd]
   14 root         0 SW   [khubd]
   15 root         0 SW   [kworker/0:1]
   16 root         0 SW<  [cfg80211]
   17 root         0 SW   [kswapd0]
   18 root         0 SW   [fsnotify_mark]
   19 root         0 SW<  [crypto]
   28 root         0 DW   [enable_swp_task]
   29 root         0 SW<  [dwc_otg]
   30 root         0 SW<  [deferwq]
   31 root         0 SW   [kworker/u2:1]
   32 root         0 SW<  [kworker/0:1H]
  255 root         0 SW   [kworker/u2:2]
  270 root         0 SW   [mmcqd/0]
  390 root         0 SWN  [jffs2_gcd_mtd5]
  403 root      149m S    ./dgiot
  404 root      1240 S    /bin/getty -L ttyS1 57600 vt100
  412 root      2008 S    ./daemon
  426 root      1240 R    telnetd
  482 root         0 SW   [RTW_CMD_THREAD]
  491 root      1116 S    wpa_supplicant -Dwext -iwlan0 -c /etc/conf/wpa_supplicant.conf
  525 root      1240 S    udhcpc -b -i wlan0 -h dgiot -s /usr/share/udhcpc/default.script
  565 root      1244 S    -sh
  708 root         0 SW   [kworker/0:2]
 1358 root      1240 R    ps
~ # mount
rootfs on / type rootfs (rw)
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
none on /sys/kernel/debug type debugfs (rw,relatime)
ramfs on /var type ramfs (rw,relatime)
ramfs on /dev type ramfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mtdblock5 on /etc/conf type jffs2 (rw,relatime)
/dev/mmcblk0p1 on /mnt/mmc/mmc1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)```

cjj25, 2022-12-15:
Ok, place this patched copy binary on your sd card.

Use `ps` to find the pid of the currently running application, for example the output shows **403**.. kill it via `kill -9 403`

Browse to your sdcard using `cd /mnt/mmc/mmc1/` and run `./dgiot.patched `

See if it starts, what's its output and if its stable, check RTSP on rtsp://{IP_ADDRESS}:8554/

[dgiot.patched.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/10237415/dgiot.patched.zip)  extract this

McPrapor, 2022-12-15:
Segmentation fault:
```/mnt/mmc/mmc1 # ./dgiot.patched
Segmentation fault (core dumped)```

I updated the link above, added there backup.7z with flash partitions dumps.

cjj25, 2022-12-15:
Try this one

[dgiot.patched.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/10237587/dgiot.patched.zip)

McPrapor, 2022-12-15:
Better, might be working. But there is another problem, if I kill dgiot process and run another one, it prints a lot of debug and device reboots after saying "===========>Middleware WifiStationM
[dgiot_log_patched.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/10237728/dgiot_log_patched.txt)

[dgiot_log_stock.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/10237727/dgiot_log_stock.txt)
odeCreate". Here I'm posting the stock dgiot output and the patched one.

cjj25, 2022-12-15:
I believe that's reconnecting to WiFi.

Can you perform the same actions but this time pipe the output to a log file.

For example ./dgiot.patched > /tmp/output.log

Then reconnect to telnet and grab the /tmp/output.log (cat /tmp/output.log) or pipe it to the sdcard

cjj25, 2022-12-15:
It could also be setting up an access point (looking at the code!)

McPrapor, 2022-12-15:
Hm, interesting, but it doesn't write to file, when I place it on sdcard. Anyway I was able to save the log to /etc/conf, it almost same, just several more lines about wifi. After that device reboots, even, if I start stock dgiot binary.
[dgiot_log_patched2.txt](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/10237968/dgiot_log_patched2.txt)

cjj25, 2022-12-15:
Yeah it's getting mixed up because it's already initialised its state (we don't want to be overwriting the flash just yet).

Here are some key locations that you may need to delete before starting the patched binary:

- rm -rf /tmp/regs
- rm -rf /tmp/tuya.log

We basically want to start the binary like it's a fresh instance / new boot.

Let me know if that allows it to start, if not we might need to overwrite the partition.

McPrapor, 2022-12-15:
I cannot see any files in /tmp. find says there is not tuya.log anywhere. I tried to kill other processes but without any luck. Is it possible to replace FS on root partition with some RW filesystem?

cjj25, 2022-12-15:
What files exist in `/etc/conf `when the stock binary is running?

We don't really want to risk flashing mtdblocks until we're happy with the binary, if we can find away of testing without that it will reduce the risk of bricking the device (or desoldering and reflashing the chip).

A really cheap hacky way of maybe forcefully killing that binary at boot would be to make an exceptionally large product.cof  file @ /etc/conf/product.cof  as that has RW on the partition to hopefully make it overflow or segfault. I've attached an example dummy file.

The only problem is you might not get your telnet daemon if its not executed before the stock config.

Make sure you do a backup first! 
[product.cof.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/10238714/product.cof.zip)

If this fails, we'll have no choice but to repack the mtdblocks and possibly make the rootfs RW.

McPrapor, 2022-12-15:
Going to try and report. 
So what I can see right now:
```
~ # find /etc/conf/
/etc/conf/
/etc/conf/tuya
/etc/conf/tuya/tuya_enckey.db
/etc/conf/tuya/log_seq_stat
/etc/conf/tuya/tuya_user.db_bak
/etc/conf/tuya/tuya_user.db
/etc/conf/product.cof
/etc/conf/Config
/etc/conf/Config/Json
/etc/conf/Config/Json/Camera.second
/etc/conf/Config/Json/NetWork
/etc/conf/Config/Json/Camera
/etc/conf/Config/Json/Detect
/etc/conf/Config/Json/Detect.second
/etc/conf/dgiot.log
/etc/conf/daemon.log
/etc/conf/custom
/etc/conf/custom/CustomConfig2
/etc/conf/asd.log
/etc/conf/wpa_supplicant.conf
```

cjj25, 2022-12-15:
Can you also provide the output of these:

- cat /etc/conf/dgiot.log
- cat /etc/conf/daemon.log
- cat /etc/conf/asd.log

cjj25, 2022-12-15:
Don't use that `/etc/conf/product.cof `  file, it'll fill the flash (dumb moment there on my behalf), do it on the sdcard.

McPrapor, 2022-12-15:
Ah, sorry, those are basically files where I wrote output of dgiot binary.

McPrapor, 2022-12-15:
Any idea what is that "daemon"? Quite small and launches together with the main binary. A watchdog?

cjj25, 2022-12-15:
There's an interesting stop check in the boot process.

If UART was functional, you could hit 'n' to break the standard boot.

I'll have to flash your firmware onto my dev cam and get a patch together.

```
read -t 1 -n 1 char
if [ "$char" == "n" ];then
	#mount -t nfs -o nolock 192.168.0.10:/home/luogn/workspace/nfs /mnt/
	echo "stop"
else
	#insmod /usr/lib/modules/mv_motor_gpio.ko
	cd /usr/bin/
	#./ota_upgrade&
	./daemon&
	./dgiot&
fi`
```

McPrapor, 2022-12-15:
Yep, saw that, pity that there is active no uart. I just checked the file, with it on sddcard it doesn't even start network.

cjj25, 2022-12-15:
I can see your uboot is compiled with 

`bootargs=console=null,57600 root=/dev/mtdblock4 rts_hconf.hconf_mtd_idx=2 rts-quadspi.channels=dual mtdparts=m25p80:8192k@0(global),256k@0k(boot),64k@256k(hconf),1792k@320k(kernel),5376k@2112k(rootfs),704k@7488k(userdata)`

Notice "**bootargs=console=null**"

McPrapor, 2022-12-15:
Well that sounds bad as well. Any luck with your hardware? Would be nice to get little bit changed init to run additional scripts directly from sdcard.

McPrapor, 2022-12-19:
So returning to rewrite, it about unpacking squashfs, packing back and writing to proper /dev/mtdblockX, not flashing it with chip removal?

McPrapor, 2022-12-19:
So I tried this https://github.com/cjj25/RTS3903N-Tools/issues/1#issuecomment-941176409
But stuck at the moment, that I cannot unload kernel modules without killind "dgiot" process. And if I kill it, device being rebooted  in 5 seconds. Anyway next I started to play with the dev/mtdblock and something went wrong, the device is not booting anymore. I'm going to remove the chip and reflash it and try some modifications of the init.

cjj25, 2022-12-19:
I spent a little bit of time trying to patch the current UBoot to enable console output no success yet.