code-explorer1234/Yi-RTS3903N-RTSPServer #31: VICTURE SC220

mjpradeepraj, 2023-03-24:
The script worked on victure sc220 with a minor change. Thank you very much. 
Backup attached. 
[Victure-SC220-backup.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/11062359/Victure-SC220-backup.zip)
[image]

Markdevine79, 2023-03-27:
Is there any possibility you could post a zip of the fat32 directory contents? I don't know what to do with that file format!
I have the same camera and have an output but it's quite pink and inverted top-bottom

mjpradeepraj, 2023-03-27:
I haven't found a way to fix the inverted video but fixed the pink output. Sdcard content in the below link.

https://we.tl/t-JbW1cSIW4o

cjj25, 2023-03-29:
Thanks for sharing @mjpradeepraj 

Re: Rotation

If you would like to play with the output rotation image, you can recompile source and [modify this line](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/7763840399c08659c6b2e875bc4bca5a30f7f58b/build/src/stream.c#L234)

mjpradeepraj, 2023-03-29:
Thanks for the information @cjj25 

I recompiled the stream source but got a segmentation fault on executing stream.

[image]

cjj25, 2023-03-29:
@mjpradeepraj  Did you compile using the Realtek toolchain?

I'm not sure which OS you're using but if you're using a Linux or macOS with Docker installed, you can just run ./compile.sh and this will spin up a docker container and compile everything for you.

mjpradeepraj, 2023-03-29:
@cjj25 I compiled from https://github.com/cjj25/Yi-RTS3903N-RTSPServer

I use Kali Linux OS and used docker to spin the container and compile. 

`┌──(kali㉿kali)-[~/Desktop/Yi-RTS3903N-RTSPServer]
└─$ sudo sh compile.sh 
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /to_build/output
[ 50%] Building C object CMakeFiles/stream.dir/src/stream.c.o
[100%] Linking C executable stream
Cleaning output directory
Fixing annoying dynamic linker issues
[100%] Built target stream
Build stream grabber complete, compiling RTSPServer
/toolchains/sdk /to_build
/to_build
--- Build environment setup ---

TOOLCHAIN_TARGET: mips-linux-uclibc
TOOLCHAIN_BIN:  /toolchains/sdk/bin

  Symlinking gcc
  Symlinking addr2line
  Symlinking ar
  Symlinking as
  Symlinking c++
  Symlinking cpp
  Symlinking g++
  Symlinking gcc
  Symlinking ld
  Symlinking nm
  Symlinking objdump
  Symlinking ranlib
  Symlinking strip

PATH=/toolchains/sdk/bin:/toolchains/sdk/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
TOOLCHAIN_TARGET:       mips-linux-uclibc
TOOLCHAIN_ROOT: /toolchains/sdk
STAGING_DIR:            /toolchains/sdk
TOOLCHAIN_BIN:  /toolchains/sdk/bin

GCC location: /toolchains/sdk/bin/gcc
GAS location: /toolchains/sdk/bin/as
GLD location: /toolchains/sdk/bin/ld
G++ location: /toolchains/sdk/bin/g++
CC location:  /toolchains/sdk/bin/cc

UTIL_STATIC:      
C_STATIC:         /toolchains/sdk/lib/libc.a
/toolchains/sdk/mips-linux-uclibc/lib/libc.a
DL_STATIC:        /toolchains/sdk/lib/libdl.a
/toolchains/sdk/mips-linux-uclibc/lib/libdl.a
PTHREAD_STATIC:   /toolchains/sdk/lib/libpthread.a
/toolchains/sdk/mips-linux-uclibc/lib/libpthread.a
STDCXX_STATIC:    /toolchains/sdk/lib/libstdc++.a
/toolchains/sdk/mips-linux-uclibc/lib/libstdc++.a
GCCEH_STATIC:     /toolchains/sdk/lib/gcc/mips-linux-uclibc/4.8.5/el/libgcc_eh.a
[... 935 lines cut ...]
./lib/libsbc.so.1
./lib/librtstream.so.2
./lib/librtsosd.so.1
./lib/libatomic.so.1
./lib/libasound.so.2
./lib/libopencore-amrnb.so.0
./wpa_supplicant_sample.conf
./invert_adc_rename_me
./Yi/
./Yi/ko/
./Yi/ko/cpld_periph.ko
./Yi/ko/ssp_ms41909.ko
./Yi/ko/pid_list.ko
./Yi/ko/ssp_ms41909_union.ko
./Yi/load_cpld_ssp
./rRTSPServer
./stream
./wifi/
./wifi/config.sh
./wifi/make_backup.sh
./wifi/fork_process.sh
                                                                                                                                  
┌──(kali㉿kali)-[~/Desktop/Yi-RTS3903N-RTSPServer]
└─$ 
`

mjpradeepraj, 2023-03-29:
@cjj25  I compiled from https://github.com/cjj25/Yi-RTS3903N-RTSPServer

I use Kali Linux OS and used docker to spin the container and compile.

```
┌──(kali㉿kali)-[~/Desktop/Yi-RTS3903N-RTSPServer]
└─$ sudo sh compile.sh 
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /to_build/output
[ 50%] Building C object CMakeFiles/stream.dir/src/stream.c.o
[100%] Linking C executable stream
Cleaning output directory
Fixing annoying dynamic linker issues
[100%] Built target stream
Build stream grabber complete, compiling RTSPServer
/toolchains/sdk /to_build
/to_build
--- Build environment setup ---

TOOLCHAIN_TARGET: mips-linux-uclibc
TOOLCHAIN_BIN:  /toolchains/sdk/bin

  Symlinking gcc
  Symlinking addr2line
  Symlinking ar
  Symlinking as
  Symlinking c++
  Symlinking cpp
  Symlinking g++
  Symlinking gcc
  Symlinking ld
  Symlinking nm
  Symlinking objdump
  Symlinking ranlib
  Symlinking strip

PATH=/toolchains/sdk/bin:/toolchains/sdk/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
TOOLCHAIN_TARGET:       mips-linux-uclibc
TOOLCHAIN_ROOT: /toolchains/sdk
STAGING_DIR:            /toolchains/sdk
TOOLCHAIN_BIN:  /toolchains/sdk/bin

GCC location: /toolchains/sdk/bin/gcc
GAS location: /toolchains/sdk/bin/as
GLD location: /toolchains/sdk/bin/ld
G++ location: /toolchains/sdk/bin/g++
CC location:  /toolchains/sdk/bin/cc

UTIL_STATIC:      
C_STATIC:         /toolchains/sdk/lib/libc.a
/toolchains/sdk/mips-linux-uclibc/lib/libc.a
DL_STATIC:        /toolchains/sdk/lib/libdl.a
/toolchains/sdk/mips-linux-uclibc/lib/libdl.a
PTHREAD_STATIC:   /toolchains/sdk/lib/libpthread.a
/toolchains/sdk/mips-linux-uclibc/lib/libpthread.a
STDCXX_STATIC:    /toolchains/sdk/lib/libstdc++.a
/toolchains/sdk/mips-linux-uclibc/lib/libstdc++.a
[... 937 lines cut ...]
./lib/librtstream.so.2
./lib/librtsosd.so.1
./lib/libatomic.so.1
./lib/libasound.so.2
./lib/libopencore-amrnb.so.0
./wpa_supplicant_sample.conf
./invert_adc_rename_me
./Yi/
./Yi/ko/
./Yi/ko/cpld_periph.ko
./Yi/ko/ssp_ms41909.ko
./Yi/ko/pid_list.ko
./Yi/ko/ssp_ms41909_union.ko
./Yi/load_cpld_ssp
./rRTSPServer
./stream
./wifi/
./wifi/config.sh
./wifi/make_backup.sh
./wifi/fork_process.sh
                                                                                                                                  
┌──(kali㉿kali)-[~/Desktop/Yi-RTS3903N-RTSPServer]
└─$ 

```

mjpradeepraj, 2023-03-29:
Stream works well on RTS_AV_ROTATION_90R and RTS_AV_ROTATION_90L but gives segmentation fault when RTS_AV_ROTATION_180 is used.

cjj25, 2023-03-29:
That's very odd and points to some incompatible library files. 

You can try looking in the /var/log outputs to see it gives any clues. I also have a compiled strace binary on my tools repo (check my profile) that you can use to start the stream eg. ./strace stream 1

This will give you all the sys calls and show the last call. There's also GDB on the tools repo for further debugging.

Markdevine79, 2023-04-23:
> Thanks. will play around.

Did you get anywhere with this by any chance?