code-explorer1234/Yi-RTS3903N-RTSPServer #32: Mibao D600

jmccarty1974, 2023-05-06:
I have a Mibao D600 running firmware version 7.1.00.25A_202002271051. I have tried several brands of sdcards from 2gb up to 16gb. Nothing seems to work. Anyone have this cam and did you get this to work? Any reply would be greatly appreciated.

cjj25, 2023-05-09:
Have tried formatting the SD card via the phone's app first?

TheLastMuri, 2023-08-24:
Hey there (:
I had no problems installing the new firmware (at least I think so). 
Afterwards it was possible to open the rtps stream with vlc player and with the stream running, getting access via telnet within cmd (but I don't get telnet access as long as the stream isn't running). 
Running PowerShell / nmap also was succesfull, telling me the telnet (23) and rtsp (dont remember rn) ports are open.

Now to the problem.. 
The Cam is doing its start up test, but not rotating the lens down, so its always filming its inside. Which means: no picture at all (okey, a pink / grey screen is possible).
Also I have no idea how to do anything with the cam within telnet commands. 
Like. rotating the camera to get a picture, whatsoever.
But I guess, the problem is, as soon as I contact the camera with telnet, it seems to pretty instantly freeze.
Don't think it's the WiFi connection, since it seems to be of a good stable quality.

Any ideas?

closed 2025-01-13 by jmccarty1974