code-explorer1234/Yi-RTS3903N-RTSPServer #33: New features ?

LeMaitre4523, 2023-05-10:
Hello, I come to the news, find out if this project is still news after 2 years

cjj25, 2023-05-10:
I've not updated the repository in a while but I'm still monitoring and answering any questions.

LeMaitre4523, 2023-05-10:
So the implementation of features in the TODO of readme is not planned

cjj25, 2023-05-10:
They're still on the TODO but when I have some spare time available

LeMaitre4523, 2023-05-10:
Okay, do you have any idea how long it’s gonna take you?

cjj25, 2023-05-10:
I have no immediate plans with my current work load. What feature do you need exactly?

LeMaitre4523, 2023-05-10:
Mainly the PTZ Control function, the rest is secondary

cjj25, 2023-05-10:
Which device do you have specifically?

LeMaitre4523, 2023-05-10:
I own an IP dome camera of the Keyke sub-brand

cjj25, 2023-05-10:
Do you usually control it via the Yi app? Yi usually has its GPIO pin map in the custom kernel module that's loaded

LeMaitre4523, 2023-05-10:
Yes absolutely, I control it with the YI IoT app

cjj25, 2023-05-10:
I can't do anything at the moment but you're welcome to provide a copy of the dump from the firmware.

Then when I do have a spare moment, I'll use your firmware as a test as most of the firmware I've used are non pan/tilt so it'll be useful to compare the kernel module.

LeMaitre4523, 2023-05-11:
I have no folder name backup, the SD card remains empty

cjj25, 2023-05-11:
Have you followed the instructions on the front page of this repo?

LeMaitre4523, 2023-05-12:
Indeed I did not understand that I had to install the hack

LeMaitre4523, 2023-05-29:
Hello, I come back here, I got back on my camera, and I still had the problem that the camera was not running the "hack" and I realized that the V03 does not have a Factory folder while the other two versions have one, is that the problem? And I was able to backup the firmware

cjj25, 2023-05-30:
The V03 uses a different payload to start the camera application. You can certainly try the earlier versions to see if that gets you up and running.

Which method did you use to backup your firmware? Could you send a copy and I'll see why the payload isn't working on V03 for you.

LeMaitre4523, 2023-05-30:
I actually tried the V02, this one worked well, there is still a problem of connection to the wifi, and I used the commands of your backup script with telnet, so I have mtd block files I know more what

LeMaitre4523, 2023-05-30:
Okay, you need all of them? I noticed that there was such a thing as userdata

cjj25, 2023-05-30:
Ideally I need mtdblock0 which contains everything, I'll remove the link once it has been submitted.

Just so you're aware, it'll contain your WiFi password and api keys for Yi (being fully transparent here).

LeMaitre4523, 2023-05-30:
Yeah it doesn’t matter, I’m sending you this tonight

LeMaitre4523, 2023-05-30:
This is for you!

https://we.tl/t-rAmbUQpU33

cjj25, 2023-05-31:
Can you confirm if this is a Tuya based camera or a Yi? What app are you using in the mobile?

LeMaitre4523, 2023-05-31:
According to my research it’s a Yi sub-brand, it’s called Keyke and I use the Yi IoT App

LeMaitre4523, 2023-05-31:
I do have access to telnet, and there is wifi, but I have nothing for rtsp

LeMaitre4523, 2023-05-31:
Yes   
 
PS:
  `PID USER       VSZ STAT COMMAND
    1 root      1428 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW   11 root         0 SW   12 root         0 SW   13 root         0 SW   14 root         0 SW   [khubd]
   15 root         0 SW   [kworker/0:1]
   16 root         0 SW   17 root         0 SW   [kswapd0]
   18 root         0 SW   [fsnotify_mark]
   19 root         0 SW   28 root         0 DW   [enable_swp_task]
   47 root         0 SW   48 root         0 SW   [kworker/u2:1]
   49 root         0 SW  302 root         0 SW   [kworker/u2:2]
  307 root         0 SWN  [jffs2_gcd_mtd5]
  386 root      1428 S    syslogd
  388 root      1420 S    klogd
  393 root      1428 S    telnetd
  397 root      1428 S    init
  467 root         0 SW   [RTW_CMD_THREAD]
  519 root       772 S    ./log_server
  520 root      9432 S    ./dispatch
  524 root      1104 S    ./mp4record
  525 root      8512 S    ./oss
  526 root     38192 S    ./rmm
  527 root      1084 S    ./watch_process
  530 root      1420 S    watchdog -t 2 -T 5 /dev/watchdog
  631 root      1072 S    ./arp_test
  919 root      1284 S    wpa_supplicant -c/tmp/wpa_supplicant.conf -g/tmp/wpa
  944 root      1428 S    udhcpc -i wlan0 -b -s /home/app/script/default.scrip
  988 root      9528 S    ./cloud
 1150 root     17956 S    ./p2p_tnp
 1667 root      1444 S    -sh
 1712 root      1428 R    ps`

DF:
`Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                    1664       236      1428  14% /
/dev/root                 3456      3456         0 100% /rom
tmpfs                    17128       544     16584   3% /dev/shm
tmpfs                       64         0        64   0% /media
/dev/mtdblock5            1664       236      1428  14% /overlay
overlayfs:/overlay        1664       236      1428  14% /
/dev/mtdblock4            1408      1408         0 100% /home`
MTD:
`dev:    size   erasesize  name
mtd0: 00800000 00010000 "global"
mtd1: 00030000 00010000 "boot"
mtd2: 00150000 00010000 "kernel"
mtd3: 00360000 00010000 "rootfs"
mtd4: 00160000 00010000 "userdata"
mtd5: 001a0000 00010000 "rootfs_backup"
mtd6: 00010000 00010000 "vd1"
mtd7: 00010000 00010000 "conf"`

cjj25, 2023-06-01:
I can see the V03 should work for your camera according to `/app/script/wifidhcp.sh` and `/app/script/ethdhcp.sh`

When you apply the files to your SD and turn on the camera, what happens? No WiFi?

LeMaitre4523, 2023-06-01:
Well I think I got screwed because of the normal start, if I look at the ports open, I see the port rtsp and I have more access on the app, so sorry the hack does work x)

cjj25, 2023-06-01:
Have you tried connecting to the RTSP port?

LeMaitre4523, 2023-06-01:
Yes and everything works perfectly

cjj25, 2023-06-01:
Great news!