code-explorer1234/Yi-RTS3903N-RTSPServer #34: Guidance on getting the OWSOO cameras to work

nabilt, 2023-06-02:
Hi,

I picked up an RTS3903N based camera from a brand called OWSOO model number XY-R9320-S3. I've tried various SD card hacks, including this one, but the backup directory is never created.  The UART pins are labeled, but it looks like they are disabled (all 0V and no activity on boot). Do you have any suggestions on a logical next step? I have attached the firmware which I got by reading the flash chip over SPI.

[Owsoo_cam_firmware.zip](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/files/11638149/Owsoo_cam_firmware.zip)

cjj25, 2023-06-04:
It looks like your camera executes `/tmp/sd/Factory/config.sh` on startup.

This is your entrypoint, have you tried the V02 on this repo as it used this entrypoint.

What would you like to achieve with this camera? 

The UART pins are disabled in the bootloader and/or toggled into GPIO pins at startup to be used by the IR sensor or PAN/TILT if needed.

nabilt, 2023-06-04:
Thanks for the response. I just tried V02, but no luck. I added a line to create a file in the root of the SD card, but it wasn't created. It makes me think I'm doing something wrong. Do I need to setup the camera with the official software for any hacks to work?

My goal is to use the camera with as little manufacturer software as possible. A complete open firmware would be nice, but an RTSP stream will do if that's not possible.

cjj25, 2023-06-05:
Have you been through the initial pairing sequence (via their app). I know there's a basic check for this on boot?

nabilt, 2023-06-12:
I got busy with work so I haven't had a chance yet.  I need to find a burner phone, because I don't want to install the app on mine. I'll report back if I make some progress.