code-explorer1234/Yi-RTS3903N-RTSPServer #38: Image rotation?

ASerbinski, 2024-01-09:
On my cameras (XY-R9320-S3), the output image is rotated by 180 degrees. Is there any simple way to correct this rotation?

cjj25, 2024-01-13:
There's not a quick option but the source code responsible for the [rotation is here.](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/7763840399c08659c6b2e875bc4bca5a30f7f58b/build/src/stream.c#L234)

Changing to RTS_AV_ROTATION_0 to RTS_AV_ROTATION_1, RTS_AV_ROTATION_2 or RTS_AV_ROTATION_3  should allow you to find the correct rotation.

jlopez77, 2024-07-01:
> Didn't work, made it crash -- no telnet even so I don't know how to debug it. Ended up setting up an instance of zoneminder and using that to do the rotation.
> 
> And before you ask, yes my builds have been successful -- I've tweaked some other settings successfully, like disabling the day/night sensor since I'm shooting through windows at night.

The parameter is h264_attr.rotation and can be set to 0 (default), 1, 2 and 3 as defined in headers/rtsavdefs.h

enum rts_av_rotation {
        RTS_AV_ROTATION_0 = 0,
        RTS_AV_ROTATION_90R = 1,
        RTS_AV_ROTATION_90L = 2,
        RTS_AV_ROTATION_180 = 3,
};

It compiles but it's not a surprise, it crashes with a core dump. 
The problem here is that all this code is super untested, copied from here and there and somehow half working. My guess is that, sadly, this repo is abandoned.

Would be great if at least @cjj25 releases the documentation and/or knowlege adquired.