code-explorer1234/Yi-RTS3903N-RTSPServer #47: Sricam SP017: stream fails to open /dev/video51

ubeF, 2024-11-29:
I tried to run the RTSPServer on a Sricam SP017, which is a different RTS3930N based IP camera. 
The server itself seems to work but __stream__ fails to set up the fifo stream in /tmp/h264_high_fifo.

Stream fails after calling rts_av_create_isp_chn(&isp_attr) and outputs "fail to create isp chn, ret = -21".
It turns out that rts_av_create_isp_chn tries to open /dev/video51 without previously creating the device.

Is the creation of /dev/video51 already taken care of by a different program in YI Cameras?

I cant find any public information on the rtscamkit library or the camera drivers, so i don't know what rts_av_create_isp_chn actually does and how the camera device is supposed to be set up. Can anyone point me in the direction of the source or some documentation for the library?

cjj25, 2024-11-29:
@ubeF it's been a while since I worked on this project.

If I remember correctly, the drivers load first which then setup the /dev/video51 instances. It could be that your libraries don't match those that were compiled with the binary.

Do you have a firmware dump?

ubeF, 2024-11-29:
I don't have a full dump, but i have the relevant files here [firmware.zip](https://github.com/user-attachments/files/17963395/firmware.zip)

/etc/init.d/rcS is run on startup, does some setup stuff, runs /rom/setup.sh and then /ipc/ipc, which handles all of the camera stuff

/rom/setup.sh does not initially exist and is writable, so i used it to start your scripts.

But it turns out that /dev/video51 is not created before /ipc/ipc is run, so stream doesn't work.
I poked around /ipc/ipc a bit and i think it might use rts_av_create_chn() to create the video device but I'm not sure what exactly that function does.

cjj25, 2024-11-30:
Yes, you're right. I remember now!

If you would like more information on that functionality [see here.](https://github.com/cjj25/Yi-RTS3903N-RTSPServer/blob/ec79b746c5093e2e08a49ffb4494a82f8bfdf09f/build/src/stream.c#L224)

cjj25, 2024-11-30:
Doing some research into this camera, it appears to support ONVIF. This means it could potentially have an RTSP server built into the IPC.

It would be interesting to see we could patch this binary to enable it by default.

ubeF, 2024-12-03:
I found the code responsible for intializing /dev/videoX in ipc and added it to [stream.c](https://github.com/ubeF/Yi-RTS3903N-RTSPServer/blob/d2fb7803a41d0fc94849027e4b01c30a0722344e/build/src/stream.c). Now i can capture raw footage from __/dev/video51__.

Stream creates __/tmp/h264_high_fifo__, but it seems to be empty. There's probably a problem with the encoding, but i don't quite understand the library functions you are using.

Any ideas on what the issue could be?

roondar, 2025-07-28:
Have you made any progress @ubeF?

ubeF, 2025-08-03:
I never got the hardware-accelerated encoding working. Decided to put the RTSP-Server on another device and stream the video there via ffmpeg. But because it's encoding everything via the CPU it's only 360x240 video at like 5 FPS